FreeCAD/FreeCAD #17907: RC4  does not start on Wayland

mendberg, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Starting FreeCAD 1.0 RC4 AppImage via GUI (Gnome) results in no visible reaction. Starting it from a terminal I get an error message

```
   AppImageLauncher FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage                                                  Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
execv error: No such file or directory

```

Previous versions did not have that issue. 

```
    uname -a
Linux blue 6.10.13-3-MANJARO #1 SMP PREEMPT_DYNAMIC Tue Oct  8 03:24:49 UTC 2024 x86_64 GNU/Linux

```

### Full version info

```shell
I can not copy the info as FC does not even start
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-16:
You can install FreeCAD 1.0 RC4 from extra repository
`sudo pacman -S freecad`

mendberg, 2024-11-16:
I'm on Manjaro, only 1.0rc1-2 is available.

Was was quite happy with AppImage so far.

tarman3, 2024-11-16:
Sorry. I thought you write about Archlinux.
Also try to download RC4 ( FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage ) and last dev (FreeCAD_weekly-builds-39215-conda-Linux-x86_64-py311.AppImage)

It terminal get message - (qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
but run without problem.

tarman3, 2024-11-16:
What is it **AppImageLauncher**?

GAZ082, 2024-11-16:
I have no issues with latest Appimage in Gnome Wayland, also Arch.

GAZ082, 2024-11-16:
Skip the Appimage launcher, run the Appimage file as a standard executable (after chmod +x it, of course)

maxwxyz, 2024-11-16:
Closing this if this solves it.

closed 2024-11-16 by maxwxyz

eblanshey, 2024-11-19:
@mendberg please see this issue for a relevant thread: https://github.com/FreeCAD/FreeCAD-Bundle/issues/331

The short version is that AppImageLauncher is outdated and doesn't work with the newest FreeCAD. You have to uninstall it. You can find alternate programs that provide AppImageLauncher's functionality. Gear Lever was suggested, and I confirm it works well with FreeCAD.

mendberg, 2024-11-19:
@eblanshey – Thanks. I've already uninstalled AppImageLauncher and solved the problem.

HoWilgh, 2024-11-19:
Can confirm that uninstalling/removing AppImageLauncher does solve the problem. I could run RC on a different ubuntu 22.04 computer before. 

A note at the download/releases site would help for sure!

Maybe keywords for the github-search help: AppImage, RC4, "execv error: No such file or directory", "./FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage"

mendberg, 2024-11-19:
Yes, this needs to be mentioned at leased in the release notes.