FreeCAD/FreeCAD #17911: Main branhc - gcmodule.c SEGFAULT

ParticleTruthSeeker, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Main git branch build segfaults at start suggesting there is a problem with the garbage collector. Using a static boost build with python3.11 on Debian 12.
```
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
update_refs (containers=<optimized out>) at ../Modules/gcmodule.c:423
423	../Modules/gcmodule.c: No such file or directory.

```

Any suggestions on how to debug would be welcome.

### Full version info

```shell
Main on Debian 12 Build
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-16:
are you pulling HEAD ?

ParticleTruthSeeker, 2024-11-16:
> are you pulling HEAD ?
> 
> More details, how are you building ? what are your cmake flags etc... ?

The git branch says main. I am building with cmake and default build flags nothing different than defaults to be honest.

luzpaz, 2024-11-16:
Whats the full path to the file ?

ParticleTruthSeeker, 2024-11-16:
> Whats the full path to the file ?
> 
> Searching github i only see this: ![Screenshot_20241116_092255](https://private-user-images.githubusercontent.com/4140247/386852825-4f7d3bda-9757-425e-b609-e1d2cf74a9d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE3NjczODEsIm5iZiI6MTczMTc2NzA4MSwicGF0aCI6Ii80MTQwMjQ3LzM4Njg1MjgyNS00ZjdkM2JkYS05NzU3LTQyNWUtYjYwOS1lMWQyY2Y3NGE5ZDMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTZUMTQyNDQxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2I5NTI4Njc0M2RiMzkzNTk3YTUwZjdkMGU2YWU1ZmEwMDg4YmFjYjQ0MjE4NjFmODcyYzM1MzY3NGM0ZTU2MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.C8w6-wmqQBwVYntW0kok8hHHFlJ1zaDaaUAOeyskTzI)

It seems to be the way pythons garbage collector is being called.
https://github.com/python/cpython/blob/main/Modules/gcmodule.c

Here is another example
https://bugs.python.org/issue13616

this is the commit hash

```
commit ed40de3e2390f11b3ce3f5951fc083f954d7d088 (HEAD -> main, origin/main, origin/HEAD)
Author: luzpaz <luzpaz@users.noreply.github.com>
Date:   Wed Oct 16 14:36:02 2024 +0000

    Fix various typos
    
    Found via codespell.

```

ParticleTruthSeeker, 2024-11-18:
Any updates on this? Is main building for most people?

luzpaz, 2024-11-20:
@ParticleTruthSeeker can you try with latest HEAD?

ParticleTruthSeeker, 2024-11-20:
> @ParticleTruthSeeker can you try with latest HEAD?

Sure, ill try as soon as I can. I run it through valgrind and gdb suggesting it was trying to read a deallocated pointer of a pytuple object.

ParticleTruthSeeker, 2024-11-20:
> @ParticleTruthSeeker can you try with latest HEAD?

So this does not help. 
these are the last lines on gdb
```
   *\n#*", ' ' <repeats 43 times>...) at /home//Freecad/FreeCAD/src/Base/Interpreter.cpp:249
#59 0x0000000005b91675 in Gui::Application::runInitGuiScript() () at /home//Freecad/FreeCAD/src/Gui/Application.cpp:2117
#60 0x0000000005d4f340 in Gui::StartupPostProcess::showMainWindow() (this=0x1ffeffcf40) at /home//Freecad/FreeCAD/src/Gui/StartupProcess.cpp:423
#61 0x0000000005d4da22 in Gui::StartupPostProcess::execute() (this=0x1ffeffcf40) at /home//Freecad/FreeCAD/src/Gui/StartupProcess.cpp:228
#62 0x0000000005b923e5 in Gui::Application::runApplication() () at /home//Freecad/FreeCAD/src/Gui/Application.cpp:2281
#63 0x0000000000127597 in main(int, char**) (argc=1, argv=0x1ffeffdc68) at /home//Freecad/FreeCAD/src/Main/MainGui.cpp:295

```

luzpaz, 2024-11-21:
> This may be a problem with qt5 build. My qt6 build loads.

This is an important point! What specific qt versions are you testing ?

ParticleTruthSeeker, 2024-11-21:
> > This may be a problem with qt5 build. My qt6 build loads.
> 
> This is an important point! What specific qt versions are you testing ?

I have both builds. The Qt6 one complains about pyside6 but otherwise loads. The Qt5 build is using a 5.15.13 which I built from source. It was working before, but I updated my MPI installation which all works fine and BOOST but now it segfaults on startup complaining about Pytuples.

luzpaz, 2024-11-21:
can you attach build log ?

ParticleTruthSeeker, 2024-11-21:
> can you attach build log ?

I can do it when I am back on the relevant PC.

ParticleTruthSeeker, 2024-11-28:
> can you attach build log ?

The memory error reported in valgrind. Pointed to my python interface which lead me to PySide2. It amounts to a memory alignment issue that I identified by compiling PySide2 with -fsanitize=address,undefined -Wcast-align. There are other errors reported many times associated with pointer type casting or accessing elements in arrays that don't exist. I fixed it by downloading the source for a patched version for my distribution which probably means I should have been using the binary all along, but I had some problems with the native qt version.

Anyway the issue is fixed for now I don't know what changed and why it broke my installation all of a sudden. There are other bugs that these compilation flags show in FreeCAD but unfortunately I don't know how to help with this write now.

luzpaz, 2025-05-03:
Is this issue still relevant ?

ParticleTruthSeeker, 2025-05-04:
Sorry it was mainly a pyside bug per my last message I can close the issue.

closed 2025-05-04 by ParticleTruthSeeker