FreeCAD/FreeCAD #17913: Assembly: Multiple instances of the same part are all affected when using Transform on one of them

angstrem98, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Placing multiple instances of the same part in assembly and then using Transform on one of them causes transform to be applied to all of them.

## Steps to reproduce
- Load 2 simple parts, let's call them Part_A and Part_B
- Create new assembly
- Insert Part_A in the assembly and make it grounded
- Insert Part_B in the assembly
- Insert another instance of Part_A
- Finish insert process
- Right click on the last added instance of Part_A and click Transform
- Transforming orientation for example causes even the grounded first Part_A to change orientation
Same applies if you have multiple instances of part B and you try transforming them.
I cannot reproduce it now, but in some cases adding more assembly constraints can trigger this behavior.
Also, changing visibility of one instances affects all of them.

## Expected behavior
All instances should behave like independent parts.

### Full version info

```shell
1.0.0RC4
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

huguesdpdn-aerospace, 2024-11-18:
## Test
@angstrem98 - Cannot reproduce - Did I reproduce it wrong ? I even tried a third "A" part ... nothing.

I put in attachment my part A and B (the "B" is the worst B I have ever made, forgive me)

## Video and file to reproduce
[PartAB.zip](https://github.com/user-attachments/files/17793950/PartAB.zip)

https://github.com/user-attachments/assets/ba33cb2f-349b-441c-a620-b310d1ab5a56

## FreeCAD version
Using latest version from weekly builds:
```
OS: Ubuntu 24.04.1 LTS (LXQt/Lubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39215 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b3a2268289482c65c18bbb4cc073b33020e71ce2
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

angstrem98, 2024-11-18:
@maxwxyz 
[asm_issue.zip](https://github.com/user-attachments/files/17796629/asm_issue.zip)
Attached you will find the demo project.
Open the asm file and activate the assembly.

@huguesdpdn-aerospace Thank you for your video. I will now provide more specific instruction to reproduce the issue.
Instead of clicking transform from the tree menu, do this:
- left click on Part_A (that is not grounded) in 3D view. A face on it will be highlighted.
- Right click on it and click Transform

This will trigger the described issue.

huguesdpdn-aerospace, 2024-11-18:
Hello,

double bug discovered! Thanks @angstrem98 . 

## Test about transformation/rotation : **CONFIRMED**
Even if the part is grounded or not:
- doing a transform from the Model tree view window => only move/rotate the selected part => **OK**
- doing a transform from a face selection in the 3D view window => move/rotate all instances of the part => **BUG**

https://github.com/user-attachments/assets/520ba61b-b326-499f-8376-048105a19927

## Now, I discovered a second issue, a SIGSEGV crash : **CONFIRMED**

When I did a `Undo`  while adding a file in the Assembly, it generates a crash:

```
17:02:59  <Exception> invalid container
17:02:59  <App> Transactions.cpp(186): Exception on undo 'Insert Component':invalid container

17:03:21  Running the Python command 'Assembly_InsertLink' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 83, in Activated
    self.panel = TaskAssemblyInsertLink(assembly, view)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 120, in __init__
    self.buildPartList()
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 241, in buildPartList
    process_objects(guiDoc.TreeRootObjects, docItem)
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 202, in process_objects
    if not obj.ViewObject.ShowInTree and not self.showHidden:
           ^^^^^^^^^^^^^^

Object has been removed from document

... *** Exiting FreeCAD *** ...

<Exception> invalid container
<App> Transactions.cpp(186): Exception on undo 'Insert Component':invalid container
Running the Python command 'Assembly_InsertLink' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 83, in Activated
    self.panel = TaskAssemblyInsertLink(assembly, view)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 120, in __init__
    self.buildPartList()
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 241, in buildPartList
    process_objects(guiDoc.TreeRootObjects, docItem)
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 202, in process_objects
    if not obj.ViewObject.ShowInTree and not self.showHidden:
           ^^^^^^^^^^^^^^

Object has been removed from documentRunning the Python command 'Assembly_InsertLink' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 83, in Activated
    self.panel = TaskAssemblyInsertLink(assembly, view)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 120, in __init__
    self.buildPartList()
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 241, in buildPartList
    process_objects(guiDoc.TreeRootObjects, docItem)
  File "/tmp/.mount_FreeCAiCpHcP/usr/Mod/Assembly/CommandInsertLink.py", line 202, in process_objects
    if not obj.ViewObject.ShowInTree and not self.showHidden:
           ^^^^^^^^^^^^^^

Object has been removed from documentProgram received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7d34de445320]
#1  /lib/x86_64-linux-gnu/libc.so.6(+0x2041c0) [0x7d34de6041c0]
```

https://github.com/user-attachments/assets/7406fa3f-e40e-4761-9398-5e6ef5a00888

huguesdpdn-aerospace, 2024-11-18:
Arg no. I recorded all my screens in the videos... sorry :/ My bad. Don't hesitate to make it full screen and focus on the right side to be able to read and see what i did.

closed 2025-11-25 by maxwxyz