FreeCAD/FreeCAD #17914: Erro de inicialização

Paulo-Pierobom, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

O Freecad 1.0 CR2  apresenta problema na inicialização,  normalmente fecha logo após a sua abertura, e quando abre apresente uma alista de erros quando usado nesta condição  trava. Estou usando um  AMD Ryzen 5 4500 6-Core Processor   3.60 GHz com 16 GB de RAM co Windows Pro  versão 22H2 Compilação do SO 19045.5131

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * A2plus 0.4.65
  * Assembly4 0.50.15
  * fasteners 0.5.24
  * Manipulator 1.5.7
  * PieMenu 1.7.5
  * Render 2024.7.4
  * sheetmetal 0.4.22
```

### Subproject(s) affected?

PartDesign

### Anything else?

10:38:39  During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in C:\Users\paulo.pierobom\AppData\Roaming\FreeCAD\Mod\fasteners\.\InitGui.py
10:38:39  Please look into the log file for further information
10:38:39  During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in C:\Users\paulo.pierobom\AppData\Roaming\FreeCAD\Mod\PieMenu\.\InitGui.py
10:38:39  Please look into the log file for further information
10:38:39  [Render][Init] Checking dependencies...
10:38:39  QObject::killTimer: Timers cannot be stopped from another thread
10:38:39  QObject::~QObject: Timers cannot be stopped from another thread
10:38:46  [Render][Init] Checking package 'PyQt6-WebEngine' - OK
10:38:46  [Render][Init] Checking package 'PyQt6' - OK
10:38:46  [Render][Init] Checking package 'QtPy' - OK
10:38:46  [Render][Init] Checking package 'wheel' - OK
10:38:46  [Render][Init] Checking package 'setuptools' - OK
10:38:47  [Render][Init] Checking package 'renderplugin' - OK
10:38:47  [Render][Init] Done.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-16:
Please specify bugs in english (i've already done it for you). 
Can you start FreeCAD in safe mode and see if the issue persists ? 

https://wiki.freecad.org/Std_RestartInSafeMode/

Paulo-Pierobom, 2024-11-16:
I updated to version CR4, but the problem persists, the link https://wiki.freecad.org/Std_RestartInSafeMode/ has no information and displays this text:
Std RestartInSafeMode/
There is currently no content on this page. You can search for this page title on other pages, or search for related records. Note that, however, you do not have permission to create this page.
On this same PC I have version 0.21 running normally.
I don't know how to make Freecad run in safe mode.

Paulo-Pierobom, 2024-11-16:
500 / 5.000
I updated to version CR4, but the problem persists, the link https://wiki.freecad.org/Std_RestartInSafeMode/ has no information and displays this text:
Std RestartInSafeMode/
There is currently no content on this page. You can search for this page title on other pages, or search for related records. Note that, however, you do not have permission to create this page.
On this same PC I have version 0.21 running normally.
I don't know how to make Freecad run in safe mode.

maxwxyz, 2024-11-16:
https://wiki.freecad.org/Std_RestartInSafeMode/en

Paulo-Pierobom, 2024-11-18:
boot into safe mode and apparently Freecad has started, what do I need to do for it to operate normally?

Paulo-Pierobom, 2024-11-18:
To start Freecad to access safe mode, I need to insist because the application opens and closes after opening for a few seconds, so I insist until it opens with a series of errors previously reported.

luzpaz, 2024-11-20:
>  I need to insist because the application opens and closes after opening for a few seconds, so I insist until it opens with a series of errors previously reported.

Can you write this differently. I don't understand what you are conveying here

Paulo-Pierobom, 2024-11-21:
Freecad does not run on my PC,

luzpaz, 2024-11-21:
@Paulo-Pierobom can you install FreeCAD on another drive beside C: ?

Paulo-Pierobom, 2024-11-21:
I have already installed Freecad1.0 on two notebooks, and it works fine on them. It just doesn't work on my desktop PC.

luzpaz, 2024-11-21:
> can you install FreeCAD on another drive beside C: ?

Notice that I asked about drive c:\ (not another machine)

Paulo-Pierobom, 2024-11-22:
I uninstalled it from drive C and installed it on drive D and the problem persists.

error relatory:

08:50:33  During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in C:\Users\paulo.pierobom\AppData\Roaming\FreeCAD\Mod\fasteners\.\InitGui.py
08:50:33  Please look into the log file for further information
08:50:33  During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in C:\Users\paulo.pierobom\AppData\Roaming\FreeCAD\Mod\PieMenu\.\InitGui.py
08:50:33  Please look into the log file for further information
08:50:33  [Render][Init] Checking dependencies...
08:50:33  QObject::killTimer: Timers cannot be stopped from another thread
08:50:33  QObject::~QObject: Timers cannot be stopped from another thread
08:50:34  [Render][Init] Checking package 'PyQt6' - OK
08:50:34  [Render][Init] Checking package 'setuptools' - OK
08:50:34  [Render][Init] Checking package 'wheel' - OK
08:50:34  [Render][Init] Checking package 'PyQt6-WebEngine' - OK
08:50:34  [Render][Init] Checking package 'QtPy' - OK
08:50:34  [Render][Init] Checking package 'renderplugin' - OK
08:50:34  [Render][Init] Done.
08:50:35  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
08:50:37  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
08:50:39  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
08:50:41  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
08:50:43  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
08:50:45  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation

luzpaz, 2024-11-22:
hey @shaise, have you seen this weirdness in the addon paths before?

> 08:50:33 During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in C:\Users\paulo.pierobom\AppData\Roaming\FreeCAD\Mod\\**fasteners.\InitGui.py**
08:50:33 Please look into the log file for further information
08:50:33 During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in C:\Users\paulo.pierobom\AppData\Roaming\FreeCAD\Mod\\**PieMenu.\InitGui.py**

luzpaz, 2025-05-10:
@Paulo-Pierobom ping

luzpaz, 2025-07-08:
No follow-up. Closing.