FreeCAD/FreeCAD #17916: UI/UX: Default Light Theme element visibility problem

semhustej, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the default Light Theme, I find three problems with visibility of elements in the 3D view. All of them happen, when faces are viewed from the **direction parallel to the normal** (top, front, etc.)of the face and **orthographic** view setting.

Steps to reproduce:

- Open FreeCAD
- Select 2D Draft as a new file (default view is top)
- Draw rectangle
- Visibility problems occur

**1)** Visibility of lines when hovering over then
When mouse is hovered over a line from the top view, the contrast between background and the "highlighted" geometry is very low. See Screenshot:
[image]

**2)** Visibility of faces
When faces are created, they are indistinguishable from the background in the top view. See screenshot:
[image]

**3)** No difference between selected and unselected face
Face not selected:
[image]

Face selected:
[image]

All of this leads to reduced overview over the workflow for the user.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Core

### Anything else?

I tried to discuss in the forum: https://forum.freecad.org/viewtopic.php?t=92078
but got no response yet. Because Classic theme will not be offered in the 1.0 release by default, roughly 1/2 users will resort to the light theme. I think it is very important to solve this issue before the 1.0 release, that is why I am raising the question here.

2 solutions came into my mind, none of them is ideal

A) change the default material to dark grey. With this one, problem 2 is solved. This also does not work well with the default Dark Theme.

B) change the default lighting direction. This solves problem 2 and 3. But it might have undesirable results in different situations, I did not test much. See example in the video (the lighting setting used is definitely not the right one, it just shows the idea):

https://github.com/user-attachments/assets/5fb6d8db-fbb8-4f69-818c-72139b3d31dd

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-16:
Just 1) is an issue of the light theme. @MisterMakerNL FYI

MisterMakerNL, 2024-11-16:
I see this issue on draft wb it's kinda annoying, classic looks a bit better. 
I though it was using the default selection color but this isn't the case. It is using #e5ffff
I tried finding the offending color tho and couldn't find it.
You got any idea @obelisk79 ?

obelisk79, 2024-11-16:
This appears to be a 3d rendering issue. Most likely a combination of spotlight angle, and material settings.

MisterMakerNL, 2024-11-16:
Makes no sense having lightning affect selection color.
This all is probably way more complicated than I understand. Maybe @kadet1090 can help us understand this, I see overlap with the highlight feature he is working on.
I would say 2D elements shouldn't be affected by lighting. And highlight and hoover colors should not change color also.

maxwxyz, 2024-11-17:
This issue is just about the selection color in Light, similar to 
- https://github.com/FreeCAD/FreeCAD/issues/15746

There is one about the shading of selection color (although IMO it should not remove shading but only shininess on selected faces):
- https://github.com/FreeCAD/FreeCAD/issues/10614

semhustej, 2024-11-17:
This does not happen only to 2D elements. It happens with faces of 3D objects too (when faces are viewed from direction parallel to the normal of the face). 
When using the default draw style, all faces are displayed based on their material combined with the lighting.

semhustej, 2024-11-17:
> If you change the material to plaster, you will see color as it has no shininess.

I don't thin that is the case. I tried changing shininess of the default material to 0% with no change in the result when viewing from the top.
What changes the result is mostly the specular color (this is the color of the light bouncing off the face). 
Changing diffuse color (that is basically the normal color of the material) also works, but that makes the material too dark in other views / dark theme.

semhustej, 2024-11-18:
I also found this discussion related to the issue 2), it has been discussed already, with no solution baked yet.
https://forum.freecad.org/viewtopic.php?t=85545

It seems to me the default light source direction should be set to 0.2,-0.2,-1.0 . This is a workaround for the issue 2), with some limitations.

But it seems to me the best solution for issue 2) would be to create a new Draw Style, which would be based on the 'Flat Lines' draw style, but would apply specular color #000000 (black) to all materials. This would reduce the white specular highlights, which are currently causing issues when looking at a face from direction parallel to the face's normal.

maxwxyz, 2025-02-27:
Closed by https://github.com/FreeCAD/FreeCAD/pull/19397

closed 2025-02-27 by maxwxyz