FreeCAD/FreeCAD #17917: FreeCad 1.0 Assy WB Max iterations on Moderate sized assemblies

jimsybeldon, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The maximum iterations limit is exceeded in moderate sized assemblies.  Iterations grow "exponentially' with joints perhaps?

Recompute after every joint added helps only to a certain extent.   Things get slower and more cpu intensive a more joints are in the model.

Concept details for the model:
- bodies are fairly primitive.  
- joints are distance,  formed by 3 orthogonal faces for each body to lock the 3 dof required.
- TNP issues are being avoid by keeping bodies simple.  

I did not use a fix joint since this would involve adding mating hole features.   The new features often induce TNP issues, which is something I needed to avoid if I were to get valid results pertaining only to the assy issues.

[image]
[image]
[image]
[image]
[image]
[image]

the model for your testing
[FirePlaceSketchHolderMAXITERATIONS.zip](https://github.com/user-attachments/files/17787556/FirePlaceSketchHolderMAXITERATIONS.zip)

Try to add body to the assembly model and add joints.    If you do not recompute after every body/joint is added, it get painfully slow.

If you do recompute after every joint it helps.  That gets less effective as you add several more bodies.

Either way, I have not gotten to 200 total joints by any workaround I could think of.      The solver eventually bogs down to use up your cpu resources.

This true of other body in other assemblies.   The attached model is simply representative of this solver issue.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.61
  * Assembly3 0.12.0
  * Assembly4 0.50.6
  * Help 1.0.3
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MiqCG, 2024-11-18:
Hi, @jimsybeldon ,
I have downloaded your file. I agree with you that FreeCAD becomes slow managing assemblies with more than N joints BUT I think this depends greatly on the type of joints. Thinks get preeeeety slow when moderately complex mechanisms (slider, revolute, cylindrical...) get in. I guess this is a known limitation.

You said "I did not use a fix joint since this would involve adding mating hole features" but you can use Fix joints just with a corner of each lap. Then you can flip the laps or apply an offset (that's a distance in the blue axis). This would greatly reduce the load on the solver as you only need one joint for each component, it is much faster to assemble and, as far as I know, the Fix joint is the least computationl demanding joint. I suggest you give it a try with this method in your project. Then I guess the "solver limit" will be much further than with 135 distance joints that there are in the example file.

[image]

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

jimsybeldon, 2024-11-20:
Hello @MiqCG

I appreciate the suggestion.    This project was done in several other cad formats with ease so doing this in FreeCad was a good way to see where things are at in FreeCad1.0.    

I was surprised how soon TNP (https://github.com/FreeCAD/FreeCAD/issues/17776) and Solver issues arose.   I was hitting issues far too early to make good use of the Assembly WB.

I believe I can find work arounds to most if not all and then help the project evolve.  I will let you know what I find.

For explanations sake:
-  This was an experiment with the most basic bodies.  Ones that do not have TNP traps.
- The require joints for 3 degrees of freedom were typical when one does not use "fixed" joints.  But that gave me the best chance to get deeper into the assembly solver.  It reveals early onset of processing issues that surprised me.
- Vertex fixed joints are an excellent why to get far deeper into a larger assembly before slowing begins.

What I wanted to learn:
-  Since I was believed Toponaming was not a significant issue,  I believed I could add fastener hole features and repeat the assembly with fixed joints at real features for the actual assembly.  (A basic good practice.)  But TNP issues were immediately a problem.   Those hole features generally have to be added once a assembly is fairly complete.  Things like fastener access can be seen at that point,  etc.

What I want to prove to myself:
- Will vertex features work for assemblies? (I believe I have seen TNP in vertex names-  I use a lookup spreadsheet to configure the body sketch dimensions for the rectangular shapes.   A change value might be renaming vertices.)
-  Can I get stable subassemblies and make a top level assembly to disperse joints among assembly levels?  Fewer each.  Then would a recompute just work at a given assembly level?  It should, but yet to be proven.
- If I can use the absolute best practices to avoid TNP issues, can I modify a part and mostly avoid exploding an assembly?  Those best practices are sort of onerous and not necessary in other cad systems.   But if it works, I can get something done and add good feedback to FreeCad after that.  Yet to be proven.
- The fastener work bench would be good to use.  But the addition of more joints might lead me to never do so.   (Fastener are great in an shop environment where everything being is the model makes producing the assembly accurate and fully described.)
-

Shkolik, 2024-12-10:
A have the same/similar issue on really small assembly... Just 4 parts/5 joints. See #18406

jimsybeldon, 2024-12-12:
Hey All,

Vertex based fixed joints did not make any improvement.   I do suspect the heuristics used for toponaming improvements are not nearly as robust where vertex naming is part of a fixed joint.  (TNP issues came up quickly)   

Furthermore, vertex fixed joints are not representative of the real geometry that actually builds a parts into an assembly.  (We use fasteners)   

I have made many different approaches to this issue and end up with the same issues.  No workaround presented itself.

All problems with slow solving and TNP issues remain as noted in previous reported issues.  @Shkolik has a similar condition using more complex joints under  https://github.com/FreeCAD/FreeCAD/issues/18406

I thought this need triage, but was reverted to "needs confirmation" by others. @maxwxyz

Nolo contendere, non compos mentis

closed 2025-09-14 by github-actions[bot]