FreeCAD/FreeCAD #17918: [PartDesign] Pocket operation on cylindrical shape invalidates shape

johnny-bit, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm working on parametric "tower" shape. I shared my problem on the forum: https://forum.freecad.org/viewtopic.php?p=792170#p792170 

I'm attaching the file with the bug present - the body shape is invalid.
[TowerTH_v2.zip](https://github.com/user-attachments/files/17787713/TowerTH_v2.zip)

### Full version info

```shell
OS: Gentoo Linux (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * Gridfinity 0.6.0
  * Render 2024.9.25
  * ExplodedAssembly
  * Pyramids-and-Polyhedrons
  * Manipulator 1.5.7
  * A2plus 0.4.68
  * fasteners 0.5.30
  * Curves 0.6.51
  * Estimate 0.1.2
  * sheetmetal 0.5.5
  * sheetmetal.backup1729712104.8273523 0.5.1 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

johnny-bit, 2024-11-18:
With the attached file - if you delete the "Pocket001" operation - the shape is completely valid and in "good state"
Now - use the "BrickCutProfileVert2" to make a pocket. Select: reversed and OK it - the shape will still be valid.
You can increase the length of the pocket to 11mm without invalidating the shape.
Once you go past 11.2mm (eg 11.3mm, 12mm and so on) the Check geometry tool reports that the shape became invalid.

tarman3, 2024-11-20:
I think this is your mistake. You get defect, because this must be a curve, not a line.
[image]
[image]

johnny-bit, 2024-11-20:
> I think this is your mistake. You get defect, because this must be a curve, not a line. 

The 2nd profile is identical with the 1st and the 1st cuts in the same way. Over at the forums I've show this issue to happen with clones. So this to me is a bit weird: why the 1st vertical cut succeeds creating no issues, while the 2nd causes issue?

I'll try making that section curved on vertical profiles to achieve the effect I want (and I'm probably doing that in sub-optimal way anyhow) thanks @tarman3 for suggesting a workaround, I'll report back whether that worked.

tarman3, 2024-11-20:
In this situation, a surface with zero area is obtained.

[image]

tarman3, 2024-11-20:
The situation is already changing for the better. Using FreeCAD a few years ago was like walking through a minefield ))

maxwxyz, 2024-11-20:
closing this as not planned.

closed 2024-11-20 by maxwxyz