FreeCAD/FreeCAD #17919: VarSet, inconsistent handling of "Prefix group name", inconsitent field ordering

bozzzzo, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

## Inconsitent handling of "Prefix group name" checkbox

When I right-click on VarSet in property view I can add properties and there is the checkbox that I can control. Default is on, which is different from default entry dialog and from expression editor dialog.
[image]

When I doubleclick on VarSet the dialog is not showing the "Prefix group name" checkbox, and it behaves like the checkbox is unchecked.
[image]

When I add to VarSet via ExpressionEditor the dialog is not showing the "Prefix group name" checkbox, and it behaves like the checkbox is unchecked.
[image]

## Inconsitent ordering of fields in different entry boxes

field order aligned on Group

| Doubleclick | Rightclick | Expression editor |
| --- | --- | --- |
| Name | Type | Variable Set |
| Group | Group | Group |
| Type | Name | New Property (same as Name) |
| Value | _missing Value_ | Info (readonly Type) |
| add another | _missing add another_ | _n/a add another_ |
| Tooltip | Documentation (same as Tooltip) | _missing Tooltip_ |
| _missing_ | Prefix group base | _missing_ |
| | | Result |
| | | Value (unnamed) |

### Full version info

```shell
OS: macOS 14.4.1
Architecture: arm64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: ProDark.qss/unset/Qt default
Installed mods: 
  * ThreadProfile 1.89.0
  * ProDarkThemePreferencePack 1.0.0
  * fcgear 1.0.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.4.72
  * lattice2 1.0.0
  * ExtremeProDark 2.9.0
  * Curves 0.6.15
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-17:
@pieterhijma FYI

furgo16, 2024-11-18:
> When I doubleclick on VarSet the dialog is not showing the "Prefix group name" checkbox, 

> When I add to VarSet via ExpressionEditor the dialog is not showing the "Prefix group name" checkbox

If I'm not mistaken, these particular points are yet another instance of this issue: https://github.com/FreeCAD/FreeCAD/issues/14633. There are two different but very similar dialogs to edit properties, and users are led to think there is only one that sometimes behaves differently.

I would suggest to consider raising the priority of that bug.

tko, 2025-01-03:
> The "Prefix group name" option is a very bad idea IMO. Please remove that.

I'm probably missing something here but I find that option to be extremely useful, I want to be able to have a property labeled plainly as "Length" under different groups and "Prefix group name" seems to be the only way make that possible.

pieterhijma, 2025-01-13:
Since we have opposing opinions, I think it is good to have a wider discussion on the forum. I created [this forum topic](https://forum.freecad.org/viewtopic.php?style=1&t=93844). @tko and @Roy-043, feel free to comment there or here as well.

closed 2025-09-13 by kadet1090

pieterhijma, 2025-12-01:
@johnsonm,

> I'm so disappointed that I missed this discussion in real time and missed my chance to contribute.

I'm sorry to hear that you missed the discussion.

> I have made extensive use of this and found it to be a wonderful affordance for reduced typing.

After the forum post, it seemed that a majority seemed to agree on the drawbacks and taking this approach.

> When it disappeared from the UI, I spent quite a while hunting around trying to find where it might been moved. It didn't occur to me that a feature regression against the prior stable release would have been introduced on a few people's aesthetic dislike.

I would advise to create a new issue or contribute to https://github.com/FreeCAD/FreeCAD/issues/19074. With all respect, I think "aesthetic dislike" is a misrepresentation of the issues associated with the group prefix, one of which is highlighted in [this post](https://forum.freecad.org/viewtopic.php?style=1&p=803899#p803899).

To be constructive, the feature is still there, but not in the "Add Property dialog". However, if you create a property like this in the Add Property dialog:

Name: Base_Length
Group: Base

or:

Name: Params_Width
Group: Params

Then you get a property Base_Length but it is shown as Length under group Base (or similar with Params).

It is not rock solid, if you change the group, strange things happen, but then again, this is a strange feature. From what I see, you should first make sure the group is defined, then you can add properties.

johnsonm, 2025-12-02:
First, thank you!

I had seen a few complaints that it offended people's sensibilities, like this one from @Roy-043:

> The "Prefix group name" option is a very bad idea IMO. Please remove that.

Since that was just an assertion without rationale, as far as I could tell it was an "aesthetic dislike".

I'm not sure I'm inferring correctly when you say "one of which is highlighted", but perhaps this one?

> A tricky aspect of properties that are prefixed with the group name, is that you cannot place the property in a different group, which is the case for properties that are not prefixed with the group name.

Since we have property renaming, it's not clear why the property rename code would not be used if you moved a property to a new group; that would feel natural to me that the usages would be renamed just as if you used an explicit property rename.

Or you might have meant something else? I see in a later post that you added this image:

[image]

But I didn't follow how that was created. Given the guards on creating duplicates, I would guess that using property rename would also address that case, but maybe I missed something.

Thank you for the workaround! I can work with that. It will be easier for me than using a plethora of VarSets. And I've jumped into #19074 and will focus further conversation there.

PR #23426 (2025-09-13): Gui: Reconcile Add Property Dialogs
This PR removes the old "Add Property" dialog in favor of the "Add Property" dialog of VarSets.

The VarSet dialog is now generic and works for any document object or document. In documents, you don't have to select a property any longer to add a new property.

Please note that other than the old "Add Property" dialog, this dialog does not support adding properties to multiple document objects. Since this was an obscure feature (see https://github.com/FreeCAD/FreeCAD/issues/23424) and allowing adding values directly makes the logic more complicated, I believ [... cut]