FreeCAD/FreeCAD #17920: Updating freecad from within appimage failed

MichielBruijn, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm running freecad RC1 and tried to update to the latest RC4 from within freecad -> edit -> preferences -> appimage -> check for updates.
A dialog box appears that there's a new release candidate available. 
I press "install update", the box dissapears and I wait and wait and wait. Nothing happens. 
Can't even close freecad. 
Had to force close it will killall freecad 

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.27
  * Defeaturing 1.2.2
  * Assembly4 0.50.15
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

superhans65, 2024-11-16:
FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage seems to have something missing. It won't run at all and the logs indicate a missing dependency. Its size of 648MB when RC2 was 802MB is also strange.

oursland, 2024-11-17:
@adrianinsaval

MichielBruijn, 2024-11-17:
> FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage seems to have something missing. It won't run at all and the logs indicate a missing dependency. Its size of 648MB when RC2 was 802MB is also strange.

You are right. It's much smaller than RC1 which ran fine.
I can't get RC4 appimage to work.
Staying with RC1 for now.

tarman3, 2024-11-17:
appimage RC4 work in Archlinux without problem

adrianinsaval, 2024-11-17:
if you have it uninstall appimagelaucher and restart your computer please https://github.com/FreeCAD/FreeCAD-Bundle/issues/331

I don't quite know why updating rc1 would hang but just FYI that will not update to rc4 but to the latest stable release, sorry it was a bug there. In general release candidates can't be updated to higher versions, you need to do a full download, sorry.

maxwxyz, 2024-11-17:
closing this then

closed 2024-11-17 by maxwxyz