FreeCAD/FreeCAD #17921: AddonManager has problems handling newer versions

davesrocketshop, 2024-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Version 4.0 of the Rocket WB requires V1.0 features so the tag '<freecadmin>1.0</freecadmin>' was added to the package.xml file. This introduced a number of unexpected behaviours

- The workbench was no longer listed in previous versions of FC such as V0.21. I would have expected them to be able to install the last compatible version. This is a problem.
- If you are able to see it (stale cache?) it will happily install despite not having the correct version. This is user reported. I haven't been able to reproduce as it doesn't show up in my list
- If you have it installed already, it will happily upgrade despite not having the correct FC version. Due to errors in loading this results in the WB not showing up in the UI. This is a bigger problem than the first one.

I have been getting user reports of this issue from a number of places, including YouTube comments, forum posts, and issues raised on the Rocket WB github

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * CfdOF 1.27.10
  * ExplodedAssembly
  * MaterialTest
  * OpenTheme 2024.9.1
  * Render 2024.1.23
  * sheetmetal 0.5.5
[/code]
```

### Subproject(s) affected?

Addon Manager

### Anything else?

Issue #17761 is probably related.
https://github.com/davesrocketshop/Rocket/issues/71

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-17:
@chennes FYI

chennes, 2024-11-17:
>     * The workbench was no longer listed in previous versions of FC such as V0.21. I would have expected them to be able to install the last compatible version. This is a problem.

The Addon Manager currently has no way of knowing what the last compatible version was. I don't know the best way of handling this. Do you have a suggestion? (@maxwxyz this is a feature request that should get its own issue, it's not going to be trivial to implement).

>     * If you are able to see it (stale cache?) it will happily install despite not having the correct version. This is user reported. I haven't been able to reproduce as it doesn't show up in my list

You can explicitly tell the Addon Manager to display Addons that don't claim compatibility with your version of FreeCAD. If you choose to do this the Addon Manager will indeed let you install/upgrade them, by design. This is supposed to be the "I know what I'm doing so shut up and do what I say" option.

>     * If you have it installed already, it will happily upgrade despite not having the correct FC version.

The current version check is being done by the display code, and doesn't have any bearing on the installation or upgrade process. Adding a version check to the upgrade process is probably straightforward.

The most straightforward path for your users (if they want to do this via the existing Addon Manager) is for you to make a git branch that supports 0.21 (or whatever version you want to support), then have users switch to Developer mode and change to that branch.

davesrocketshop, 2024-11-17:
> The Addon Manager currently has no way of knowing what the last compatible version was. I don't know the best way of handling this. Do you have a suggestion? (@maxwxyz this is a feature request that should get its own issue, it's not going to be trivial to implement).

I've been thinking about this a little, but nothing that won't require backports. One option is to include a list of backward compatible versions in package.xml? Something like:

`<compatible freecadmin="0.21">3.3.0</compatible>`

Or instead of the version, the Tag name?

> The most straightforward path for your users (if they want to do this via the existing Addon Manager) is for you to make a git branch that supports 0.21 (or whatever version you want to support), then have users switch to Developer mode and change to that branch.

I've been giving people that as a workaround