FreeCAD/FreeCAD #17924: Changing fillet radius messes up CAM operations

chaotix-, 2024-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When changing the radius of a fillet I get a lot of errors (see below for the first one). Afterwards the the features are missing from CAM operations and I cannot add them again. Also the tool controller drop down list is empty and I cannot select a tool controller for this CAM operation anymore. Furthermore the undo operation does not recover from this problem. All I can do is delete the affected CAM operations and readd them.

To reproduce:
 1. Open this file: [Garderobe-prefillet.zip](https://github.com/user-attachments/files/17790120/Garderobe-prefillet.zip)
 2. Change the fillet radius to 5mm.

Afterwards most of my CAM operations are affected and I have to delete them and redo everything:
[image]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.30
  * A2plus 0.4.68
  * Assembly4 0.50.15
```

### Subproject(s) affected?

CAM

### Anything else?

```
12:03:50  <PropertyLinks> PropertyLinks.cpp(447): Garderobe_prefillet#Profile002.Base missing element reference Garderobe_prefillet#Clone ;Face1;:H51b,F;:H4f3:8,F.Face1
12:03:50  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAfIHjno/usr/Mod/CAM/Path/Op/Base.py", line 527, in onChanged
    if prop == "Base" and self.sanitizeBase(obj):
                          ^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAfIHjno/usr/Mod/CAM/Path/Op/Base.py", line 709, in sanitizeBase
    o.Shape.getElement(sub)
<class 'ValueError'>: {'sclassname': 'N4Base10ValueErrorE', 'sErrMsg': 'Invalid shape name ?Face1', 'sfile': '/home/conda/feedstock_root/build_artifacts/freecad_1731546400676/work/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 2019, 'sfunction': 'Part::TopoShape Part::TopoShape::getSubTopoShape(const char*, bool) const', 'swhat': 'Invalid shape name ?Face1', 'btranslatable': False, 'breported': False}
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-23:
Hi @jbaehr can you reproduce this ?

J-Dunn, 2025-07-14:
Yes. This is not specific to fillets but fillets add new faces and edges and removing them renumbers everything in sight and destroys all path ops'  Base Geometry settings. 

Classic Topo cluster bomb.

tarman3, 2026-04-24:
I have solution which not resolve the toponaming problem, but at least exclude Python errors and do Task panel usable
So user can remove from base list incorrect shapes and set new 
- #27559