FreeCAD/FreeCAD #17925: CAM: Problems with deburr path / Wrong section of circle is cut.

chaotix-, 2024-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I want to deburr a couple of edges. If I select all of them I get a traceback (see below). If I select a specific subsection of the edges the generated path cuts along the wrong section of the circle.

Steps to reproduce the error message:
Open the following file: [Garderobe-deburr.zip](https://github.com/user-attachments/files/17790159/Garderobe-deburr.zip)

To generate the error message below select the following edges and generate a deburr operation:
[image]

To generate the wrong path section of a round edge select the following edges and generate a deburr operation:
[image]
This generates the following path:
[image]

With the following edges the path is correct (except the not selected edge is missing):
[image]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.30
  * A2plus 0.4.68
  * Assembly4 0.50.15
```

### Subproject(s) affected?

CAM

### Anything else?

12:29:11  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAfIHjno/usr/Mod/CAM/PathScripts/PathUtils.py", line 70, in new_function
    res = function(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAfIHjno/usr/Mod/CAM/Path/Op/Base.py", line 788, in execute
    result = self.opExecute(obj)
             ^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAfIHjno/usr/Mod/CAM/Path/Op/Deburr.py", line 380, in opExecute
    wire = PathOpUtil.offsetWire(w, base.Shape, offset, True, side)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAfIHjno/usr/Mod/CAM/Path/Op/Util.py", line 287, in offsetWire
    owire = orientWire(wire.makeOffset2D(offset), True)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: {'sclassname': 'N4Part18NullShapeExceptionE', 'sErrMsg': 'makeOffset2D: result of offsetting is null!', 'sfile': '/home/conda/feedstock_root/build_artifacts/freecad_1731546400676/work/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 2773, 'sfunction': 'Part::TopoShape& Part::TopoShape::makeElementOffset2D(const Part::TopoShape&, double, Part::JoinType, Part::FillType, Part::OpenResult, bool, const char*)', 'swhat': 'makeOffset2D: result of offsetting is null!', 'btranslatable': False, 'breported': False}
12:29:11  Deburr002: {'sclassname': 'N4Part18NullShapeExceptionE', 'sErrMsg': 'makeOffset2D: result of offsetting is null!', 'sfile': '/home/conda/feedstock_root/build_artifacts/freecad_1731546400676/work/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 2773, 'sfunction': 'Part::TopoShape& Part::TopoShape::makeElementOffset2D(const Part::TopoShape&, double, Part::JoinType, Part::FillType, Part::OpenResult, bool, const char*)', 'swhat': 'makeOffset2D: result of offsetting is null!', 'btranslatable': False, 'breported': False}

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2026-01-31:
I assume it's still relevant

tarman3, 2026-02-03:
`makeOffset2D()` can not create offset for this wire

> <class 'ValueError'>: makeOffset2D: result of offsetting is null!

[open_profile_test (1).zip](https://github.com/user-attachments/files/25032341/open_profile_test.1.zip)

[image]

```
import FreeCAD
import Part

obj = FreeCAD.ActiveDocument.getObject("Body")
names = ("Edge1", "Edge5", "Edge7")
edges = obj.getSubObject(names)
wire = Part.Wire(edges)
Part.show(wire)
offsetWire = wire.makeOffset2D(offset=2, join=0, openResult=False)
Part.show(offsetWire)
```