FreeCAD/FreeCAD #17926: Changes to attachment of box primitive cause infinite hang

geckolinux, 2024-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi there, first of all many thanks and congratulations for the progress of FreeCAD with it being close to a 1.0 release.

I somehow managed to create a part based on a box primitive with some subsequent hole and pocket features despite the original box being improperly attached. When I try to the change the parameters of the box it won't update, and it throws an error about "infinite shape provided". So I'm trying to change the way the box is attached, but at that point FreeCAD goes into an infinite loop and hangs with CPU and RAM skyrocketing. I have repeated this all 5 or 6 times that I tried to change the attachment references.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

PartDesign

### Anything else?

[infinite.zip](https://github.com/user-attachments/files/17790768/infinite.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-17:
I think the problem is different. When you fix the attachment a recompute occurs. The size of the box has also changed, and as a result a quarter segment of a 90mm M1 threaded hole has to be computed. This results in an object with ca. 1600 faces (which is a lot).

A 90mm M1 hole seems unrealistic to me, a quarter of such a hole even more so, but I am not a mech person.

If you uncheck the `Threaded` property of the hole before fixing the attachment there is no issue.

Roy-043, 2024-11-18:
To fix the model:
1. Change the `Threaded` property of the hole to false.
2. Fix the attachment of the box.
3. Remove external geometry from Sketch002. Edges from the threaded hole were referenced here (seems strange).
4. Resize the box?

In the attached [file](https://github.com/user-attachments/files/17805121/infinite-roy.FCStd.zip) steps 1-3 have been completed (remove the .zip extension).

closed 2025-09-14 by github-actions[bot]