FreeCAD/FreeCAD #17930: Sketcher Offset Tool Creates Under-Constrained Construction Lines

Relsek-eng, 2024-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the sketcher offset tool and selecting "Add Offset Constraint" it creates construction lines that aren't fully constrained.
This leaves the sketch showing as under-constrained. My image is a basic example that can be fixed by adding coinsident constraints to each line, but this would be a real issue for complicated shapes. These constraints should be added by the offset feature.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.62
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

huguesdpdn-aerospace, 2024-11-17:
Hello @Relsek-eng , I didn't understand at all how to reproduce the issue. Can you please:

1. Execute FreeCAD
2. Open a new file
3. Go to the top bar menu
4. Click on the `Macro` menu
5. Select submenu `Macro recording...`
6. Enter `17930_Sketch`
7. Click on `Record`
8. Select the `Part Design` workbench
9. Click on the button `Create a new body` (blue piece icon)
10. Click on the button `Create a sketch` (white paper with red square and circle)
11. Recreate the sketch until the offset issue
12. Do your offset command that do not work
13. Go on the top bar and Click on the `Macro` menu
14. Click on the `Stop macro recording`
15. Go on the top bar and click on the `Macro` menu
16. Click on the `Macros...`
17. Select `17930_Sketch.FCMacro`
18. Click on the `Edit...` button
19. Copy everything
20. Go back to this current Github issue page
21. Paste everything as reponse

Thanks to this, we would be able to reproduce the same steps in the Sketch workbench and check what is the error

Relsek-eng, 2024-11-18:
# -*- coding: utf-8 -*-

# Macro Begin: C:\Users\kkesler\AppData\Roaming\FreeCAD\Macro\17930_Sketch.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import PartDesign
import PartDesignGui
import Sketcher

# Gui.runCommand('Std_DlgMacroRecord',0)
### Begin command Std_New
App.newDocument()
# App.setActiveDocument("Unnamed")
# App.ActiveDocument=App.getDocument("Unnamed")
# Gui.ActiveDocument=Gui.getDocument("Unnamed")
# Gui.activeDocument().activeView().viewDefaultOrientation()
### End command Std_New
### Begin command PartDesign_Body
App.activeDocument().addObject('PartDesign::Body','Body')
App.ActiveDocument.getObject('Body').Label = 'Body'
# Gui.activateView('Gui::View3DInventor', True)
# Gui.activeView().setActiveObject('pdbody', App.activeDocument().Body)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection(App.ActiveDocument.Body)
App.ActiveDocument.recompute()
### End command PartDesign_Body
# Gui.Selection.addSelection('Unnamed','Body')
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed','Body','Origin.')
# Gui.runCommand('Std_ToggleVisibility',0)
# Gui.runCommand('PartDesign_CompSketches',0)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed','Body','Origin.XY_Plane.',-23.5717,-32.5,0)
App.getDocument('Unnamed').getObject('Body').newObject('Sketcher::SketchObject','Sketch')
App.getDocument('Unnamed').getObject('Sketch').AttachmentSupport = (App.getDocument('Unnamed').getObject('XY_Plane'),[''])
App.getDocument('Unnamed').getObject('Sketch').MapMode = 'FlatFace'
App.ActiveDocument.recompute()
# Gui.getDocument('Unnamed').setEdit(App.getDocument('Unnamed').getObject('Body'), 0, 'Sketch.')
# import Show
# ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
# tv = Show.TempoVis(App.ActiveDocument, tag= ActiveSketch.ViewObject.TypeId)
# ActiveSketch.ViewObject.TempoVis = tv
# if ActiveSketch.ViewObject.EditingWorkbench:
#   tv.activateWorkbench(ActiveSketch.ViewObject.EditingWorkbench)
# if ActiveSketch.ViewObject.HideDependent:
#   tv.hide(tv.get_all_dependent(App.getDocument('Unnamed').getObject('Body'), 'Sketch.'))
# if ActiveSketch.ViewObject.ShowSupport:
#   tv.show([ref[0] for ref in ActiveSketch.AttachmentSupport if not ref[0].isDerivedFrom("PartDesign::Plane")])
# if ActiveSketch.ViewObject.ShowLinks:
#   tv.show([ref[0] for ref in ActiveSketch.ExternalGeometry])
# tv.sketchClipPlane(ActiveSketch, ActiveSketch.ViewObject.SectionView)
# tv.hide(ActiveSketch)
# del(tv)
# del(ActiveSketch)
# 
# ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
# if ActiveSketch.ViewObject.RestoreCamera:
#   ActiveSketch.ViewObject.TempoVis.saveCamera()
#   if ActiveSketch.ViewObject.ForceOrtho:
#     ActiveSketch.ViewObject.Document.ActiveView.setCameraType('Orthographic')
# 
# Gui.Selection.clearSelection()
# Gui.runCommand('Sketcher_CreateRectangle_Center',0)
ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')

lastGeoId = len(ActiveSketch.Geometry)

geoList = []
geoList.append(Part.LineSegment(App.Vector(-7.500000, -10.000000, 0.000000),App.Vector(7.500000, -10.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(7.500000, -10.000000, 0.000000),App.Vector(7.500000, 10.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(7.500000, 10.000000, 0.000000),App.Vector(-7.500000, 10.000000, 0.000000)))
[... 58 lines cut ...]
conList2.append(Sketcher.Constraint('PointOnObject',11,1, 7))
conList2.append(Sketcher.Constraint('PointOnObject',11,2, 1))
conList2.append(Sketcher.Constraint('Perpendicular',12, 8))
conList2.append(Sketcher.Constraint('PointOnObject',12,1, 8))
conList2.append(Sketcher.Constraint('PointOnObject',12,2, 2))
App.getDocument('Unnamed').getObject('Sketch').addConstraint(conList2)
del conList2

conList = []
conList.append(Sketcher.Constraint('Parallel',8, 2))
conList.append(Sketcher.Constraint('Parallel',5, 3))
conList.append(Sketcher.Constraint('Parallel',6, 0))
conList.append(Sketcher.Constraint('Equal',10, 9))
conList.append(Sketcher.Constraint('Parallel',7, 1))
conList.append(Sketcher.Constraint('Equal',11, 9))
conList.append(Sketcher.Constraint('Equal',12, 9))
conList.append(Sketcher.Constraint('Distance',9, 5))
App.getDocument('Unnamed').getObject('Sketch').addConstraint(conList)
del conList

App.getDocument('Unnamed').getObject('Sketch').movePoint(9,2,App.Vector(-7.500000,18.491549,0),0)
App.ActiveDocument.recompute()
# Gui.Selection.addSelection('Unnamed','Body','Sketch.Vertex18',-12.5,18.4915,0.014,False)
# Gui.Selection.clearSelection()
# Macro End: C:\Users\kkesler\AppData\Roaming\FreeCAD\Macro\17930_Sketch.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++

huguesdpdn-aerospace, 2024-11-19:
Hello @Relsek-eng ,

awesome job, thank you. Can you confirm this is what you are describing as an issue?

## Bug
This is a **CONFIRMED BUG**

## Test
1. Create a new document
2. Create a `Body`
3. Create a `Sketch`
4. Select any plan (`XY` for instance)
5. Add a centered rectangle (centered or not from the center point of the plan makes no difference)
6. Add an horizontal constraint width (whatever value)
7. Add a vertical constraint height (whatever value)
8. Select all lines/verteces/points with the mouse (selecting everything in the sketch)
9. Click on the `Offset geometry` tool icon and maintain left click
10. Press the `J` key on your keyboard
11. Move your mouse INSIDE the rectangle you created in the step 5
12. Release the left click

## Output:
**EXPECTED**: The new offset line is fully constrainted
**GOT**: The new offset is NOT fully constrained - 4 DoF(s)

## Illustration:
[image]

## Error analysis:
Missing 4 coincident constraints (construction line inner points) are not coincident with the corner points of the newly created rectangle

## Deeper analysis:
The issue is that the construction lines points are not coincident with the new rectangle corner points. There are constraints about the fact that the point should be on the line of the newly created rectangle but there is no position constraint from the newly rectangle corners/end of construction line.

The correct behavior should be that `Sketcher Offset` shall NOT create a Coincident constrait from point to the line but rather a point to point constraint.

Even worst, after creating the newly rectangle, you cannot fix this issue without deleting the 4 point on line Constraints, otherwise the solver fails.

## Additional info:
From official documentation : [Wiki FreeCAD Sketcher Offset](https://wiki.freecad.org/Sketcher_Offset), nothing defined if the `Sketcher Offset` tool should or not keep the same amount of DoFs before/after usage.

## OS / FreeCAD info:
```
OS: Ubuntu 24.04.1 LTS (LXQt/Lubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39215 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b3a2268289482c65c18bbb4cc073b33020e71ce2
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

Relsek-eng, 2024-11-19:
Hi @huguesdpdn-aerospace,
That all looks correct except some minor differences in the process from step 9 on:

9. Click on the Offset geometry tool icon **~~and maintain left click~~ (does not require me holding down the click)**
10. Press the J key on your keyboard
11. Move your mouse _INSIDE_ the rectangle you created in the step 5. **This is true if mode is left as "Arc", which seems to fully define when _OUTSIDE_ of the shape. However, changing the Mode (M) to "Intersection" leads to this issue for both _INSIDE_ and _OUTSIDE_.**
12. **~~Release the left click~~ Accept the offset by either a) type a dimension then press Enter or b) left click and release.**

closed 2026-09-08 by maxwxyz