FreeCAD/FreeCAD #17931: Learning FreeCad from Youtube video but crashes when launching "PartDesign"  Workspace

gsohler, 2024-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When Iaunch "Part Design" from the combo box, the app crashes;.
I use Windows10 here

### Full version info

```shell
I just downloaded 
FreeCAD-0.21.2-WIN-x64-installer-1.exe
```

### Subproject(s) affected?

None

### Anything else?

I searched for related issue in google and found that similar issues from 2022 are relatred to the graphics driver

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

huguesdpdn-aerospace, 2024-11-17:
Hello @gsohler,

sorry but, you know, this kind of report won't help us at all. It's like going to the doctor just saying "I feel pain.". I think you could understand that the doctor is expecting from you to describe the pain, how to do you feel it, where it was, how, when, etc. The doctor needs a lot more info to understand what is going on. So let's do that together!

Can you please:
1. Provide the YouTube video link (so we can "reproduce the same steps" if you did some before)
2. Execute FreeCAD, do NOT select `PartDesign`, do NOT select anything. Just go on the top bar and click on the `Help` menu. Then select submenu `About FreeCAD` . Click on `Copy to clipboard` and paste the output here
3. Now let's try to crash your FreeCAD program and get the output .
-   Go to your Windows desktop
- Type CTRL+R
- In the window, type `cmd` and press enter
- And now execute command: `"C:\Program\ Files\FreeCAD 0.21\bin\FreeCAD.exe" > FreeCAD_17931_Logs.txt 2>&1`
- After FreeCAD opens, select the PartDesign workbench in the list
- Wait FreeCAD to crash and close by itself (if FreeCAD do not close after 2 minutes (so wait a bit !!), close it manually or use the Taskmanager to kill the FreeCAD process)
- Select the files in your User folder (CTRL+E) =>   `C:\` => `Users` => `YourWindowsUserName` and find the file `FreeCAD_17931_Logs.txt` and attach it to your response

So in your response, we expect:
1. YouTube link
2. Full output of the help window
3. Log file `FreeCAD_17931_Logs.txt`

Once we get that, we will be able to start investigations

Thank you

gsohler, 2024-11-18:
Hi Huguesdpdn,

Thank you for your instant reply.

I understand that it is difficult  to debug an issue without  much
information given, but when I found the "crash" yesterday, the freecad
window was literally disappearing after 2 sec without any  further
actions. Sorry I missed to describe that.

Today when I tried to create the video, it was not  crashing anymore until
I noticed that Freecad only crashes for the 2nd instance on my desktop. and
until i managed to create the YouTube video,
it even displayed a nice message window! Looks like its a short-of-memory
issue ?

I know it's very difficult to debug crashes for windows applications as i
deploy windows software myself, but i am happy to learn in parallel.

In any case, please find the requested data.  I am sure, especially with
the log file provided you will be able to fix it.

https://youtu.be/FjBn0bAmF6o

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

[28496:25384:1118/084232.993:ERROR:cache_util_win.cc(21)] Unable to move the cache: Access is denied. (0x5)
[28496:25384:1118/084232.993:ERROR:cache_util.cc(139)] Unable to move cache folder C:\Users\guenther.sohler\AppData\Roaming\FreeCAD\webdata\persistent\GPUCache to C:\Users\guenther.sohler\AppData\Roaming\FreeCAD\webdata\persistent\old_GPUCache_000
[28496:25384:1118/084232.993:ERROR:disk_cache.cc(184)] Unable to create cache
[28496:25384:1118/084232.993:ERROR:shader_disk_cache.cc(606)] Shader Cache Creation failed: -2

gsohler, 2024-11-18:
Hi again,

I was not aware, that Chromium Browser  is bundled with FreeCAD. is it ?
Whats the advantage of having a  "static" Browser bundled with FreeCad
instead of relying to the users' Browser which apparently "worked"  during
downloading freecad ?
I am very new  to testing FreeCad. Can you provide me a dedicated download
link/download page where i shall test FreeCad instead ?

When look into the log file, which I delivered today, I don't recognize any
relation to chromium browser. Can you share information which manifests
your suspicion ?

And even the problem *was* chromium, I don't understand why a misbehaviour
of a subprogram could crash all FreeCad.
Isn't it possible to check for nullptr andor catch an exception with
try-catch.
The log file clearly  shows code pointers and i could try to find them in
the guthub repo ;)

BTW: I have seen some traces that freecad made big version jump 1.0 today ?
is that true/worth testing ?

xtemp09, 2024-11-19:
> Whats the advantage of having a  "static" Browser bundled with FreeCad
> instead of relying to the users' Browser which apparently "worked"  during
> downloading freecad ?

Different users on different platforms use different browsers. Instead of supporting all of them, _Qt_ is bundled with _Chromium_. The same goes to PDF ― instead of supporting all PDF viewers, _Qt_ is bundled with its own.

> I am very new  to testing FreeCad. Can you provide me a dedicated download
> link/download page where i shall test FreeCad instead ?

Release candidate: https://blog.freecad.org/2024/11/14/freecad-1-0-release-candidate-4-is-out/
Weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

> When look into the log file, which I delivered today, I don't recognize any
> relation to chromium browser. Can you share information which manifests
> your suspicion ?

There were _Chromium_ errors concerning cache stuff at the bottom of your previous post.

> The log file clearly  shows code pointers and i could try to find them in
> the guthub repo ;)

From my point of view, the log clearly shows _Chromium_ failing to acquire access to cache. Internet recommends deletion of the cache folders and restarting the program.

luzpaz, 2024-11-20:
@huguesdpdn-aerospace I'm not sure the need to troubleshoot 0.21.x related issues. Although your instructions rock and are super thought out, best way forward IMHO is to ask for update to latest stable or weekly.

closed 2024-11-20 by luzpaz

gsohler, 2024-11-20:
sorry for my late  reply.
I tested again on 1.1 dev version and on 1.0
I canNOT see the problem anymore. it appears solved.I think it got fixed by making "Part Design" default on startup.

thank you for your support!