FreeCAD/FreeCAD #17937: TNP documentation

yorikvanhavre, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a transcript of an issue that has been posted on the FPA meeting agenda, we felt it's more appropriate here. It essentially boils down to building documentation of the TNP mitigation system, what works, what does not, what can a user do, etc. Below is the original text:

With the mitigations for the topological naming problem now in place and version 1.0 of FreeCAD approaching, can a developer who liaises with members of the tutorial-making community suggest that they produce tutorials on how to deal with the TNPs that remain in FreeCAD 1.0? It would be especially helpful for novice users who recently learned how to handle the TNP prior to version 1.0, but will now have to re-learn how to handle the TNP for version 1.0.

Users will need to know the situations in which the TNP has been eliminated in version 1.0 and the situations in which the TNP remains.

Users will also need to know what techniques to use in FreeCAD 1.0 to deal with the situations where the TNP continues to exist and which of the pre-version 1.0 techniques used to deal with the TNP are no longer needed in version 1.0.

Before FreeCAD 1.0, users would either use a datum plane with an offset or use a sketching plane with an offset. Which of these two techniques will still be needed in FreeCAD 1.0, and in which situations?

### Full version info

```shell
m/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-18:
who to tag on that issue?

CalligaroV, 2024-11-18:
@yorikvanhavre @maxwxyz for me it's fine if you tag/assign this issue to me.
I'd say that this makes sense ATM as I'm mostly following issues related to TNP.

Regarding the OP, by now I can share some resources:
1. The original documentation about the parts of the algorithm, from RealThunder's repo: https://github.com/realthunder/FreeCAD_assembly3/wiki/Topological-Naming
2. Notes taken by bgbsww and me in the early days of work on the import of the code for the FPA Grant: https://github.com/FreeCAD/DevelopersHandbook/issues/92
3. The actual (Nov 18th, 2024) list of issues, and therefor cases, fixed by the TNP code imported/developed until now (coarse, but may help to check if a particular case is already handled): https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue+is%3Aclosed+label%3A%22Topic%3A+Toponaming%22
4. The descriptions of the various methods imported are in the `.h` files (`Toposhape.h`, `ComplexGeoData.h`, `ElementMap.h`, ...)

Providing a complete list of what's possible to do right now is something I think should be done with the contributions of other experienced users, mainly those who give support on the forum and on the other platforms.

ATM for what I remember I think I can say that probably the Pad/Pocket operations are the ones that are handled better.
Other operations may not be that robust yet.

If you have any questions don't hesitate to ask!

maxwxyz, 2024-11-18:
@FEA-eng @Roy-043 maybe for the wiki