FreeCAD/FreeCAD #17940: Basic_Part_Design_Tutorial_Example_017.fcstd loads with error in Pocket001.

DougSmithShop, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was going to do the Basic Techdraw Tutorial, but when I loaded the Basic_Part_Design_Tutorial_Example_017.fcstd I got an error.  I tried on RC2 and the Weekly Build 39215 with the same errors.  The file **_does_** load properly in 0.21.2
[image]
[image]
[image]

### Full version info

```shell
OS: fossapup64 9.5 (xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-18:
@Roy-043 can you look into the wiki? https://wiki.freecad.org/Basic_TechDraw_Tutorial

DougSmithShop, 2024-11-18:
Do you want me to construct the model as instructed in the Wiki using RC4?
That would not solve the problem of being instructed in the "Basic TechDraw
Tutorial" that instructs the user to first download the referenced file.

Roy-043, 2024-11-19:
This is really a TNP issue. In the model sketches are attached to faces. In V1.0 some faces have a different index, which breaks the model.

https://wiki.freecad.org/Basic_Part_Design_Tutorial_019 should result in a more stable model. I'll try to upload that model and then update https://wiki.freecad.org/Basic_TechDraw_Tutorial.

Roy-043, 2024-11-19:
Actually https://github.com/FreeCAD/Examples/tree/master seems unmaintained. Not sure what should be done.

Roy-043, 2024-11-19:
Try using the model attached to this forum topic:
https://forum.freecad.org/viewtopic.php?t=73235&start=10#p649493

DougSmithShop, 2024-11-19:
The file "pd_tut_019_roy.FCStd" loads correctly in both RC4 and weekly_39240.  Could this part replace the one referenced in the various tutorials?

Roy-043, 2024-11-21:
I have created a related issue:
https://github.com/FreeCAD/Examples/issues/11

ipatch, 2026-01-19:
i just recreated the file using the latest dev build, as i wanted to test some PRs from today's PR merge meeting. here is the file i created from working through the basic part design tutorial on the wiki. feel free to share and modify the file __to your heart's content__ all you need to do is unzip the file.

[wiki.basic.part.design.tutorial.final.zip](https://github.com/user-attachments/files/24723651/wiki.basic.part.design.tutorial.final.zip)

[image]

[image]

luzpaz, 2026-09-11:
Should be addressed before next release. Upping priority.

luzpaz, 2026-09-12:
@maxwxyz any chance a fix for this can get squeezed into #32622 ?

maxwxyz, 2026-09-12:
The file is not part of the source files so I see no need to change something.

luzpaz, 2026-09-12:
My bad. I missed that it's wiki related. Cancel clear.

Roy-043, 2026-09-13:
This not just a documentation problem. The file in question is a valid v0.17 file that breaks in more recent versions (also in v0.21) due to TNP issues.