FreeCAD/FreeCAD #17943: Copy/Cut/Paste is grayed out in sketcher

cajhne, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Copy Cut and Paste commands do not function in the latest appimage of FreeCAD on Ubuntu 24.04 LTS.
Not only do the hotkeys not work, but also, the items in the Edit Menu for those functions are greyed out.

Steps to reproduce:
1. Open FreeCAD
2. Make new part, and new sketch on XY plane
3. make circle
4. select circle
5. Choose Edit > Copy, then Edit > Paste

What should have happened:
FreeCAD should copy the selected sketcher geometry to clipboard, and then pasted a copy back to sketcher

What happened instead:
FreeCAD functions for Cut/Copy/Paste are greyed out in the Edit menu.

Thanks for your help!

### Full version info

```shell
[code]
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-18:
Hotkeys work in the 1.0 RC and weekly versions and the command is called "Copy... , Cut..., Paste... in Sketcher"

closed 2024-11-18 by maxwxyz

cajhne, 2024-11-18:
Confirmed, fixed in 1.0 RC4