FreeCAD/FreeCAD #17952: BIM/Draft: Commands are not ended when file is closed in the middle of command

semhustej, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- Open new file
- Start Arch_Wall command
- Close the file by clicking the X icon on the file tab
- Do not save file
- Arch_Wall is still active on the Task panel

Alternatively continue:

- Close the Arch_Wall command in the Task panel
- Open new file
- Start any command
- The command fails, error shows up:

```
17:17:33  Running the Python command 'Draft_Circle' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftguitools\gui_arcs.py", line 68, in Activated
    super().Activated(name=self.featureName)
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftguitools\gui_base_original.py", line 262, in Activated
    super().Activated(name)
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftguitools\gui_base_original.py", line 104, in Activated
    App.activeDraftCommand.finish()
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

'Arch_Wall' object has no attribute 'finish'
```

https://github.com/user-attachments/assets/15dd13d4-a8e3-48cf-9e4b-9b79751cd9c3

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

This seems to happen to all Draft and BIM tools. I tested Arch_Wall, Arch_Column, Draft_Line, Draft_Rectangle.
I tried with two commands from the Part Design WB and they seem to close properly.

I could reproduce the issue with FC 0.21.2 (with a small variation, the behavior is not exactly the same), it does not seem like a regression to me.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-11-18:
This problem is just very very very huge. It's not just about Draft/Bim.
An open tool remains on the panel in very many cases. You can switch to another document or workbench, cancel the action (ctrl+Z), a thousand other actions. But if you did not close the toolbar with the "OK" or "Cancel" button - it will be shown in the task manager.
This is very illogical behavior.

semhustej, 2024-11-19:
Hi @kaiwas 

**The first video:**
It looks like an Esc key not working with Fillet tool. That might be worth reporting as a separate issue, what do you think @maxwxyz ?

**The second video:**
Looks like not ending the command is happening when switching between projects. It seems to happen to not only BIM/Draft tools. This looks like a separate issue to closing the file, which I reported. How should we proceed @maxwxyz ?

**The third video:**
Looks like the same case as the second video, only with BIM command instead of Part command.

kaiwas, 2024-11-19:
> It looks like an Esc key not working with Fillet tool. That might be worth reporting as a separate issue, what do you think

The Esc key does not work everywhere if a click was made on the 3D window. That is, the taskbar is not considered a priority in this case.

I can file an issue for each of these cases, if necessary.
But there is a chance that there will be a lot of such cases)
I found these three examples without trying to look for them. I just performed some actions almost at random and got "problems". I am sure that there are many more options for reproducing problems (you just need to set a goal to identify them)

PR #21073 (2025-05-05): BIM: close task panels on doc close
Fixes #17952.

This PR is a continuation of #20719.

It mainly involves adding `self.doc`. This new attribute is then also used in the rest of the modified classes. In some cases code was slightly reordered for consistency.

The `_finish_command_on_doc_close` function has been modifief to also handle `Gui.Snapper.ui.escape` and the code in gui_arcs.py has been updated accordingly.

Additionally:
Removed the `done` and `spacedone` variables in BimWall.py by tweaking the code.