FreeCAD/FreeCAD #17953: Draft: Add 'Use Print Color' column to the Layers Manager

semhustej, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

User is able to set up Print Color in the Layers Manager, but he/she has no overview whether it is being turned on or off.

Proposed solution:
Add 'Use Print Color' column to the Layers Manager with the value of the property of each layer

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-11-25:
Makes sense indeed

maxwxyz, 2026-11-07:
This is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/d2eed465c67d722cee1c4701421b6708e949eb58

The Layers manager has a Use Print Color column to show and change that setting.