FreeCAD/FreeCAD #17956: RC4 AppImage does not support SpaceMouse via spacenavd

mendberg, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

### 

I've removed AppImageLauncher from my system to make the RC4 AppImage run. It now works on Manjaro Gnome, but FC no longer reacts on input from my 3Dconnexion SpaceMouse; spacenavd is running as usual.

With AppImageLauncher and previous FC versions, my SpaceMouse was working fine. The SpaceMouse still works when connected to a different computer still using AppImageLauncher, so it is not the hardware.

### Full version info

```shell
OS: Manjaro Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.47
  * fasteners 0.5.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-18:
This may not be related to AppImageLauncher.
Try to check spacemouse in RC2 without AppImageLauncher.

mendberg, 2024-11-18:
Even RC2 does ignore my SpaceMouse now. (the other computer–wehere everything is working fine–  is still running RC2 via AppImageLauncher)

adrianinsaval, 2024-11-18:
I can't reproduce the issue, is spacenavd service running in your computer?

mendberg, 2024-11-19:
@adrianinsaval Yes, it is running. I posted a line of the log just 1 comment earlier:

`Log: Connected to spacenav daemon `

It's a bit weird: I even rebooted the computer trying to fix this late yesterday. Tomorrow morning, after having the machine powerless over the night, FC and spacenavd work happily together again. -> Issue closed.

Thanks for all your support.

closed 2024-11-19 by mendberg