FreeCAD/FreeCAD #17957: Add Better 3MF Support

liberodark, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,

I cannot open any files from : https://makerworld.com/

Best Regards

### Full version info

```shell
1.0 RC2 and RC4
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-11-18:
Can you point to a specific file that doesn't work for you so I don't have to guess which one you mean? (I don't like guessing, it takes too much time)

cbagwell, 2024-11-19:
@liberodark post this example link on discord:  https://makerworld.com/en/models/798080?designId=798080#profileId-737503

I quickly looked at Freecad code and contents of a few Maker World 3MF's and looks like Freecad expects:

```
 <resources>
  <object>
   <mesh>
   </mesh>
  </object>
 </resources>
```

While maker world has this which references a different file that does have above layout:

```
<resources>
  <object>
   <components>
    <component p:path="/3D/Objects/object_42.model" objectid="1" p:UUID="002a0000-b206-40ff-9872-83e8017abed1" transform="1 0 0 0 1 0 0 0 1 0 0 0" />
   </components>
  </object>
</resources>
```

PR #17986 (2024-11-28): Mesh: Improve 3MF format to support files from 'Make World'
Fixes #17957
closed 2024-11-28 by wwmayer