FreeCAD/FreeCAD #17958: Difficulty navigating workbenches with a lot of buttons

SpeeQz1, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When it comes to beginners who are being introduced to FreeCAD, some may struggle with the large quantity of buttons being shown all at once especially for workbenches like `FEM` or `BIM`. So there needs to be devised of a way of reducing the overall clutter on the screen for each workbench while still making it easy to find the appropriate actions.

One suggestion is to implement customizable dropdown buttons like "Create an additive primitive", where a user can replace any of large amount of buttons all with a single dropdown button where they will be placed.
With this suggestion also comes the addition and implementation of more dropdown buttons overall for each workbench to declutter the UI.

[image]
[image]

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.29 (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * CurvedShapes 1.0.13
  * FreeCAD-Ribbon-Develop 1.4.0
  * SearchBar 1.0.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Mannshoch, 2024-11-19:
Is it meant to create dropdowns like on android where you drag an Icon onto an other icon and you get a group and vice-versa. 

Or more that way that you have a "Draft snap" group like in your image and somehow you could collapse it into a drop down menu?

I would like both :-)

SpeeQz1, 2024-11-19:
> Is it meant to create dropdowns like on android where you drag an Icon onto an other icon and you get a group and vice-versa.
> 
> Or more that way that you have a "Draft snap" group like in your image and somehow you could collapse it into a drop down menu?
> 
> I would like both :-)

I was hoping in general there was any way of making user defined dropdown buttons through the settings, but this visual way you are suggesting also sounds nice.

Currently such an idea is planned to be added to the Ribbon UI addon.

maxwxyz, 2024-11-20:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/9926

closed 2024-11-20 by maxwxyz