FreeCAD/FreeCAD #17961: freecad under arm

ClubACNews, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,
I use Kicad under ARM with Windows 11 for ARM with raspberry pi 4B and it's good.
Is it possible in the future to have a Freecad for ARM under Windows 11 ?
Thanks

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.29
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-11-18:
Is this a duplicate of https://github.com/FreeCAD/FreeCAD/issues/17181 ?

ClubACNews, 2024-11-18:
Certainly if the same version is also capable of working under ARM64 of a raspberry-pi and not of an The Apple M1/M2 as in the post