FreeCAD/FreeCAD #17962: Position is not respected when exporting to dxf

winstonford, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Problem: 
---
Placement -> Position is not respected when exporting to dxf from certain versions of Freecad.

### Full version info

```shell
About info for each test listed below
```

### Subproject(s) affected?

None

### Anything else?

Test:
---
1. In sketcher create a sketch and add a square. 
1. Create another sketch with a circle, inside the square. 
1. Select the circle sketch, in the Data panel in Property View > Placement > Position, set X to 2. - The circle moves to the right and outside of the square.
1.  Select both sketches and export to dxf.
1.  Open dxf in Librecad, Inkscape, Sharecad:
   - The circle is back inside the square.  Placement is not respected.  Fail: ❌ 
   - The circle remains outside the square.  Placement is respected.  Pass: ✅

Test files: 
[outside-the-box.zip](https://github.com/user-attachments/files/17808734/outside-the-box.zip)

Test file in Freecad
---
[image]

Exported dxf in Librecad
---
| Fail ❌ | Pass ✅ |
|-----|-----|
| 
[image]
| [image] |

Test Results:
---
Table below shows from which version the dxf was *exported*.
 Results seemingly not affected by which version *created* the test file.

|                           |1.1.0dev.39240 | 1.1.0dev.39215| 1.0.0RC4  |  1.0.0RC3 | 1.0.0RC2 | 0.21.2 |
| --------------- | -------------- | -------------- | --------- | --------- | -------- | ------- |
| Linux arch64    |                          |                          |                   |                  |                |              |
| Linux x86_64   |                          |                          |                   |                   |                |              |
| macOS arm64  | [^1]                 | [^1]                  |  ❌            |                   | ❌          | ✅       |
| macOS x86       | ❌                 | ❌                     |                    |                  |                | ✅        |
| Windows           |                        |✅[^2]              |                    |                  |                |               | 

 
[^1]: When attempting to open this version, I get error"“FreeCAD” is damaged and can’t be opened. You should move it to the Trash."
[^2]:  This appears to be a Pass.  ___ would you confirm? https://github.com/FreeCAD/FreeCAD/issues/12278#issuecomment-2480449443

About info for each test
---
<details>
<summary> Linux</summary>
</details>

<details>
<summary> macOS arm64</summary>
<details>
<summary>1.0.0RC4</summary>

OS: macOS 14.6.1
Architecture: arm64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: https://github.com/FreeCAD/FreeCAD/commit/83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods:

Behave-Dark-Colors 0.1.1
InventorLoader 1.4.0
Dracula 0.0.5
sheetmetal 0.5.3
Help 1.0.3
</details>

<details>
<summary>1.0.0RC2</summary>

OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Light/Fusion
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * InventorLoader 1.4.0
  * OpenTheme 2024.9.1
  * Dracula 0.0.5
  * sheetmetal 0.5.5
  * Help 1.0.3
</details>
<details>
<summary>0.21.2</summary>
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: https://github.com/FreeCAD/FreeCAD/commit/b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods:

Behave-Dark-Colors 0.1.1
InventorLoader 1.4.0
Dracula 0.0.5
sheetmetal 0.5.3
Help 1.0.3
</details>
</details>

<details>
<summary> macOS x86</summary>

<details>
<summary>1.1.0dev.39240</summary>

OS: macOS 12.7.6
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Dracula.qss/unset/Qt default
Installed mods: 
  * Dracula 0.0.5

</details>

<details>
<summary>1.1.0dev.39215</summary>

OS: macOS 12.7.6
Architecture: x86_64
Version: 1.1.0dev.39215 (Git) Conda
Build type: Release
Branch: main
Hash: b3a2268289482c65c18bbb4cc073b33020e71ce2
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Dracula.qss/unset/Qt default
Installed mods: 
  * Dracula 0.0.5

</details>
<details>
<summary>0.21.2</summary>

OS: macOS 12.7.6
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Dracula 0.0.5
</details>
</details>

<details>
<summary>Windows</summary>
<details>
<summary> 1.1.0dev</summary>

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39176 (Git) Conda
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.14
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.3
  * Silk 0.1.5
  </details>
</details>

---

This issue originally posted as a comment in a similar issue when exporting to STL:
https://github.com/FreeCAD/FreeCAD/issues/12278#issuecomment-2477794477

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-23:
This is such a great bug report. I just wanted to let you know. 😄

winstonford, 2024-11-24:
Why thank you.  I'm secretly hoping for test takers so I can add results to the table.

tarman3, 2024-11-25:
Cannot confirm.
Maybe need some option in DXF export?

[image]

<details>
  <summary>FreeCAD 1.1.0dev.39240 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

winstonford, 2024-11-25:
Thanks. Added to results.  Great idea to include export settings.  Included it after the about info and will add it for the other tests.

tarman3, 2024-11-25:
Take my words back. Before I create sketch by myself. And it looks ok.
I repeat test with your file and test failed.

And now I can't repeat a successful test.
I don't understand what I did differently. My head is spinning ))

tarman3, 2024-11-25:
Start from the beginning.
The result is different if switch on **Use legacy Python exporter**

[image]

[image]

<details>
  <summary>FreeCAD 1.1.0dev.39240 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

winstonford, 2024-11-25:
No problem. Updated the table.  Would you mind running the test in 0.21.2 in arch linux?  I thought my passing tests with 0.21.2 on mac may have been the result of using the legacy exporter, but I just checked, and legacy exporter is NOT checked. When I did check the box to use legacy exporter, I was prompted to download software, and I declined, to simplify for now.

tarman3, 2024-11-26:
Can not start FreeCAD 0.21.2 appimage in Archlinux

<details>
  <summary>Errors</summary>
(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
(qt.glx) qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
(qt.glx) qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)

Could not initialize GLX
</details>

tarman3, 2024-11-26:
Tryed FreeCAD 0.21.2 from ppa:freecad-mainteiners/freecad-stable in **LinuxMint** 22 Live
Test passed with and without legacy exporter.

<details>
  <summary>FreeCAD 0.21.2.33771 (Git), Linux Mint 22 (X-Cinnamon/cinnamon)</summary>

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
```
</details>

[image]

tarman3, 2024-11-26:
Last dev
with Legacy Python exporter: Test passed 
without Legacy Python exporter: Test not passed

<details>
  <summary>FreeCAD 1.1.0dev.39315 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51

```
</details>

winstonford, 2024-11-27:
With the last dev, I had the same sequence you did, where the first test passes and subsequent tests fail.  So it is a thing, and now I am testing 3 times.  For each test I create a new file in a new directory.

winstonford, 2024-11-27:
For anyone following, just discovered a clue. When test fails and position is *not* respected, I get the user notification: "QPixmap::scaled: Pixmap is a null pixmap"
When test passes and position *is* respected, I do not get the above notification.

winstonford, 2024-11-27:
@maxwxyz would you mind doing another test in Windows on this DXF issue?  I believe your first test passed, but we have had 2 instances now where the first test passed and subsequent tests failed.

tarman3, 2024-11-27:
> When test fails and position is _not_ respected, I get the user notification: "QPixmap::scaled: Pixmap is a null pixmap"

I dont have this message than test failed.

winstonford, 2024-11-27:
Ok, would you confirm that user notifications are enabled?

Preferences > General > Notification Area

Settings
- [x] Enable Notification Area
- [x] Enable non-intrusive notifications

Then on my screen the icon and a number displays bottom right to the left of the navigation menu.  I only noticed it after I turned on safe mode in the last dev ( top menu Help > Restart in safe mode. ) This enables both check boxes above by default.  

In 21.2, where my test passes, in top menu Help, there is no "Restart in safe mode".   A workaround I read in the forums suggests backing up user.cfg and then deleting it.  On restart a new user.cfg is created with default settings.  However, in my case, in  Preferences > General > Notification Area the first option "Enable Notification Area" was *not* checked while the second "Enable non-intrusive notifications" was checked.  I had to check the first option in order to see the notifications icon display in bottom right.

tarman3, 2024-11-28:
Yes, <ins>Enable Notification Area</ins> and <ins>Enable non-intrusive notifications</ins> is enabled.

tarman3, 2024-11-28:
Maybe the export DXF problem is related to the geometry kernel and need to pay attention not to the FreeCAD version, but to the OCC version.

luzpaz, 2025-09-02:
@winstonford has this improved at all on latest  dev build ? https://github.com/FreeCAD/FreeCAD/releases

maxwxyz, 2026-11-07:
There is another fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/0537d35d068a66765d684a512d09a08d2ce8767e

It keeps sketch placement in the projection used by the C++ DXF exporter. Unlike https://github.com/FreeCAD/FreeCAD/pull/29494 it changes the projection instead of adding offsets in the writer.