FreeCAD/FreeCAD #17964: Segfault when trying to move an assembly

dawagner, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Upon trying to drag an assembly that has two revolute joints and a belt joint, FreeCAD 1.0RC4 crashed.

Here are the FCStd files for the parts and assembly: [belt bug.zip](https://github.com/user-attachments/files/17811125/belt.bug.zip)

Here are the logs: 

```
FreeCAD 1.0.0, Libs: 1.0.0RC4R39104 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
3.8e-08 <Assembly> AssemblyObject.cpp(176): Solve failed: Gear radius is zero.
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
4.11483 <Assembly> AssemblyObject.cpp(176): Solve failed: Gear radius is zero.
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
4.13172 <Assembly> AssemblyObject.cpp(176): Solve failed: Gear radius is zero.
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
4.14024 <Assembly> AssemblyObject.cpp(176): Solve failed: Gear radius is zero.
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
7.05888 <Assembly> AssemblyObject.cpp(176): Solve failed: Gear radius is zero.
Unhandled std exception in ViewProvider::eventCallback: Gear radius is zero.
(Event type: SoLocation2Event, object type: AssemblyGui::ViewProviderAssembly)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3e9a0) [0x7fa826a5c9a0]
#1  0x7fa770c00339 in MbD::ConstraintSet::constraintsDo(std::function<void (std::shared_ptr<MbD::Constraint>)> const&) from /tmp/.mount_FreeCAJDMAej/usr/lib/./libOndselSolver.so.1+0x59
#2  0x7fa770ccb8ca in MbD::Joint::jointForceI() from /tmp/.mount_FreeCAJDMAej/usr/lib/./libOndselSolver.so.1+0xea
#3  0x7fa770ccb93e in MbD::Joint::aFX() from /tmp/.mount_FreeCAJDMAej/usr/lib/./libOndselSolver.so.1+0x1e
#4  0x7fa770b86fa2 in MbD::ASMTConstraintSet::updateFromMbD() from /tmp/.mount_FreeCAJDMAej/usr/lib/./libOndselSolver.so.1+0x72
#5  0x7fa770b777cd in MbD::ASMTAssembly::updateFromMbD() from /tmp/.mount_FreeCAJDMAej/usr/lib/./libOndselSolver.so.1+0xdd
#6  0x7fa770b7f8b7 in MbD::ASMTAssembly::runDragStep(std::shared_ptr<std::vector<std::shared_ptr<MbD::ASMTPart>, std::allocator<std::shared_ptr<MbD::ASMTPart> > > >) from /tmp/.mount_FreeCAJDMAej/usr/lib/./libOndselSolver.so.1+0x517
#7  0x7fa7e91d7c09 in Assembly::AssemblyObject::doDragStep() from /tmp/.mount_FreeCAJDMAej/usr/lib/AssemblyApp.so+0x3f9
#8  0x7fa7ddf45ced in AssemblyGui::ViewProviderAssembly::tryMouseMove(SbVec2s const&, Gui::View3DInventorViewer*) from /tmp/.mount_FreeCAJDMAej/usr/lib/AssemblyGui.so+0x35d
#9  0x7fa7ddf466ec in AssemblyGui::ViewProviderAssembly::mouseMove(SbVec2s const&, Gui::View3DInventorViewer*) from /tmp/.mount_FreeCAJDMAej/usr/lib/AssemblyGui.so+0x1c
#10  0x7fa829fb46a0 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/libFreeCADGui.so+0x400
#11  0x7fa8264d346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0xbe
#12  0x7fa8262d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0x68
#13  0x7fa82648cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0xea
#14  0x7fa8264dc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0x58
#15  0x7fa8264f713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0x2a
#16  0x7fa8262d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0x68
#17  0x7fa82648cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0xea
#18  0x7fa8264dc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0x58
#19  0x7fa8264f713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0x2a
#20  0x7fa8262d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0x68
#21  0x7fa8262df26c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0xac
#22  0x7fa8262d61ba in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0x2ea
#23  0x7fa8264a8ad2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/./libCoin.so.80+0x72
#24  0x7fa829f3714e in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/libFreeCADGui.so+0x1e
#25  0x7fa829f42618 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/libFreeCADGui.so+0x58
#26  0x7fa829f4c360 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/libFreeCADGui.so+0x40
#27  0x7fa829f5534f in Gui::TouchpadNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/libFreeCADGui.so+0x2df
#28  0x7fa829f4e265 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/libFreeCADGui.so+0x65
#29  0x7fa829f77c3b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/libFreeCADGui.so+0x6b
#30  0x7fa829f2d9e5 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAJDMAej/usr/bin/../lib/libFreeCADGui.so+0x1a5
[... 32 lines cut ...]
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * freecad.gears 1.3.0
  * Curves 0.6.48
```

### Subproject(s) affected?

None

### Anything else?

I was actually trying to reproduce a different issue with belts but this crash is preventing me from doing so.

I am a very newbie user, and new to CAD overall, and this is my first bug report.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-11-19:
Can this be a duplicate of https://github.com/FreeCAD/FreeCAD/issues/17797 ?

maxwxyz, 2024-11-20:
Cannot reproduce on Windows. Where do you drag and how?
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.31
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
´´´

luzpaz, 2024-11-20:
@dawagner i addition to elaborations requested above, please also update to [1.0.0](https://github.com/FreeCAD/FreeCAD/releases/tag/1.0.0) or [weekly](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

huguesdpdn-aerospace, 2024-11-20:
## Test

Tested on latest weekly build, both file produce no crash nor segfault.
The assembly file is working fine and moving parts in the assembly works fine. I was able to turn both parts as I wished

https://github.com/user-attachments/assets/536125ed-1ac7-467a-a803-14a2d86de607

## OS / FreeCAD info:
```
OS: Ubuntu 24.04.1 LTS (LXQt/Lubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

## Result
- no crash
- no segfault
- no Python error
- only multiple:
`<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch`
I was able to move parts inside the assembly, to drag elements in them, even move the assembly from one file to another, nothing. Cannot reproduce.

nmschulte, 2024-11-23:
> Upon trying to drag an assembly that has two revolute joints and a belt joint, FreeCAD 1.0RC4 crashed.
> 
> Here are the FCStd files for the parts and assembly: [belt bug.zip](https://github.com/user-attachments/files/17811125/belt.bug.zip)

The FCStd files contain no Belt type Joint, just two Revolutions.  I think this must be the wrong example file.

luzpaz, 2024-11-23:
@dawagner 👆

MiqCG, 2024-12-03:
Waiting for dawagner to provide the file with the Belt joint, I obtain this error
`3.7e-07 <Assembly> AssemblyObject.cpp(165): Solve failed: Gear radius is zero.`
if I add the Belt joint by myself into the original poster file.
Then, if I accept the joint creation panel and try to drag the assembly, a crash occurs.
Maybe the title could be changed to something more specific to Belt joints.

Attached the file with the attempt of Belt joint (belt-bug-assemblyMOD.FCStd): 
[belt bug Modified.zip](https://github.com/user-attachments/files/17995115/belt.bug.Modified.zip)

https://github.com/user-attachments/assets/6f9a35aa-a68e-42a0-b577-066c04f847ed

```
./FreeCAD_weekly-builds-39416-conda-Linux-x86_64-py311.AppImage
FreeCAD 1.1.0, Libs: 1.1.0devR39416 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
3.7e-07 <Assembly> AssemblyObject.cpp(165): Solve failed: Gear radius is zero.
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
55.6155 <Assembly> AssemblyObject.cpp(165): Solve failed: Gear radius is zero.
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
55.6644 <Assembly> AssemblyObject.cpp(165): Solve failed: Gear radius is zero.
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
55.6864 <Assembly> AssemblyObject.cpp(165): Solve failed: Gear radius is zero.
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
57.3779 <Assembly> AssemblyObject.cpp(165): Solve failed: Gear radius is zero.
Unhandled std exception in ViewProvider::eventCallback: Gear radius is zero.
(Event type: SoLocation2Event, object type: AssemblyGui::ViewProviderAssembly)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7f209fe5b050]
#1  0x7f2039000339 in MbD::ConstraintSet::constraintsDo(std::function<void (std::shared_ptr<MbD::Constraint>)> const&) from /tmp/.mount_FreeCAMJbhlc/usr/lib/./libOndselSolver.so.1+0x59
#2  0x7f20390cb8ca in MbD::Joint::jointForceI() from /tmp/.mount_FreeCAMJbhlc/usr/lib/./libOndselSolver.so.1+0xea
#3  0x7f20390cb93e in MbD::Joint::aFX() from /tmp/.mount_FreeCAMJbhlc/usr/lib/./libOndselSolver.so.1+0x1e
#4  0x7f2038f86fa2 in MbD::ASMTConstraintSet::updateFromMbD() from /tmp/.mount_FreeCAMJbhlc/usr/lib/./libOndselSolver.so.1+0x72
#5  0x7f2038f777cd in MbD::ASMTAssembly::updateFromMbD() from /tmp/.mount_FreeCAMJbhlc/usr/lib/./libOndselSolver.so.1+0xdd
#6  0x7f2038f7f8b7 in MbD::ASMTAssembly::runDragStep(std::shared_ptr<std::vector<std::shared_ptr<MbD::ASMTPart>, std::allocator<std::shared_ptr<MbD::ASMTPart> > > >) from /tmp/.mount_FreeCAMJbhlc/usr/lib/./libOndselSolver.so.1+0x517
#7  0x7f2062ad1235 in Assembly::AssemblyObject::doDragStep() from /tmp/.mount_FreeCAMJbhlc/usr/lib/AssemblyApp.so+0x465
#8  0x7f20600440cd in AssemblyGui::ViewProviderAssembly::tryMouseMove(SbVec2s const&, Gui::View3DInventorViewer*) from /tmp/.mount_FreeCAMJbhlc/usr/lib/AssemblyGui.so+0x35d
#9  0x7f2060044acc in AssemblyGui::ViewProviderAssembly::mouseMove(SbVec2s const&, Gui::View3DInventorViewer*) from /tmp/.mount_FreeCAMJbhlc/usr/lib/AssemblyGui.so+0x1c
#10  0x7f20a33c4d30 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_FreeCAMJbhlc/usr/bin/../lib/libFreeCADGui.so+0x400
#11  0x7f209f8d346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCAMJbhlc/usr/bin/../lib/./libCoin.so.80+0xbe
#12  0x7f209f6d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAMJbhlc/usr/bin/../lib/./libCoin.so.80+0x68
#13  0x7f209f88cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAMJbhlc/usr/bin/../lib/./libCoin.so.80+0xea
#14  0x7f209f8dc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAMJbhlc/usr/bin/../lib/./libCoin.so.80+0x58
[... 17 lines cut ...]
#32  0x7f20a117039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAMJbhlc/usr/bin/../lib/libQt5Widgets.so.5+0x6e
[... cut ...]
#50  0x7f20a02bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAMJbhlc/usr/bin/../lib/libQt5Core.so.5+0x82
#51  0x7f20a2ecb0dd in Gui::Application::runApplication() from /tmp/.mount_FreeCAMJbhlc/usr/bin/../lib/libFreeCADGui.so+0x89d
#52  /tmp/.mount_FreeCAMJbhlc/usr/bin/freecad(+0x78b7) [0x5581b5dcd8b7]
#53  /lib/x86_64-linux-gnu/libc.so.6(+0x2724a) [0x7f209fe4624a]
#54  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f209fe46305]
#55  /tmp/.mount_FreeCAMJbhlc/usr/bin/freecad(+0x7c2a) [0x5581b5dcdc2a]
```

closed 2024-12-09 by chennes