FreeCAD/FreeCAD #17965: Caliper

TrFrMa, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Create  a simple object and use the caliper to measure something, anything. A small box shows up with information about the X-Y-Z coordination. Just click in the box and freeCAD is abruptly closed. Even in safe-mode.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39176 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Qt default
Installed mods: 
  * CommandPanel
  * A2plus 0.4.67
  * CurvedShapes 1.0.9
  * Curves 0.6.47
  * offline-documentation 1.0.0-alpha
  * slic3r-tools
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-19:
Cannot confirm on 
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.31
  * freecad.gears 1.2.0
  * Manipulator 1.5.7

```

Please provide an example file

tarman3, 2024-11-19:
Confirm in RC4 and weekly-builds-39215.
Create body and sketch in PartDesign.
Open tool measurement.
Click in zone with measurements.
FreeCAD closed immediately.

maxwxyz, 2024-11-19:
@tarman3 please add full about info

tarman3, 2024-11-19:
Done

maxwxyz, 2024-11-19:
Does is also happen in latest weekly?
@hlorus FYI, maybe this is a Linux specific issue

wwmayer, 2024-11-19:
Confirmed!

tarman3, 2024-11-19:
Confirmed with FreeCAD_weekly-builds-39240-conda-Linux-x86_64-py311.AppImage

OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * MeshRemodel 1.10.34
  * freecad.gears 1.3.0
  * fasteners 0.5.30
  * sheetmetal 0.5.5
  * PieMenu 1.9.5 (Disabled)
  * Curves 0.6.51

wwmayer, 2024-11-19:
It's a typical use-after-free error.

When the user clicks the COG object then `SoFCUnifiedSelection::setSelection` is called where the view provider of the COG object is accessed. Then it calls `Gui::Selection().addSelection` where the selection observer `TaskMeasure::onSelectionChanged` is called. Because the COG object cannot be measured it will be removed and destroyed immediately which makes the pointer to the view provider inside `SoFCUnifiedSelection::setSelection` a dangling pointer. Further down in that function the pointer will be dereferenced and thus causes UB that usually ends with a crash.

PR #17969 (2024-11-22): Core: Fix dangling pointer in SoFCUnifiedSelection::setSelection
This fixes #17965: Measurement: Crash when clicking on COM label
commit be162217 (2024-11-22): Core: Fix dangling pointer in SoFCUnifiedSelection::setSelection

This fixes #17965: Measurement: Crash when clicking on COM label
commit 08fef740 (2024-11-22): Core: Fix dangling pointer in SoFCUnifiedSelection::setSelection

This fixes #17965: Measurement: Crash when clicking on COM label
closed 2024-11-22 by wwmayer