FreeCAD/FreeCAD #17966: RC4 - integrated assembly-wb - adding Fillet breaks assembly

HoWilgh, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are already many issues raised related to the integrated assembly-wb, but often complex models are used which makes it cumbersome to understand why the issue happens.
I tried to build a as simple assembly as possible to show the problem of crashing the assembly by adding simple features to one of the bodies. Here I did choose the way to simple set the tip to a prior feature to make it easier to reproduce. Adding a new feature like a fillet result in the same behavior.
I do not know if it is intended to work that way. 

### Intended behavior
PD-features like a fillet can be added to one of the bodies in the assembly and joints stay intact.
### Actual behavior
PD-feature fillet is added to one of the bodies in the assembly and prior used elements in joint cant be found any more.

### Steps to reproduce
- open file FCRC4_integratedAssemblyWB.FCStd (attached as .zip, first screenshot)
- Switch to Assembly-wb by double-clicking 'Assembly' in Tree-View
- Refresh (Ctrl-R)
- Doubleclick 'Revolute' joint -> Both edges are correctly selected
- Select 'Cancel'
- Refresh (Ctrl-R)
- Double-click the second body 'Body should be centered inside the Pocket of Body001'
- Now in PD-wb select 'Fillet001 (if set as tip the Revolute-Joint breaks)' and right-click and select 'Set tip' 
 -> Given messages in Report view:
```
08:55:34  <PropertyLinks> PropertyLinks.cpp(447): FCRC4_integratedAssemblyWB#Joint.Reference1 missing element reference FCRC4_integratedAssemblyWB#Body Body.;Edge2;:H173,E.Edge2
08:55:34  <PropertyLinks> PropertyLinks.cpp(447): FCRC4_integratedAssemblyWB#Joint.Reference1 missing element reference FCRC4_integratedAssemblyWB#Body Body.;Edge2;:H173,E.Edge2
```
-> Position of second body is already wrong! But no indication for missing information/edges is given in the model tree!

- Switch back to Assembly-wb
- Refresh (Ctrl-R)
- Doubleclick 'Revolute' joint -> Edge2 cant be found

-> Selecting 'Solve Assembly' now moves the second body further away each time when selected (See second screenshot).

Should this be covered by the TNP-fixes?

[FCRC4_integratedAssemblyWB.zip](https://github.com/user-attachments/files/17811629/FCRC4_integratedAssemblyWB.zip)
[image]
[image]

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly3 0.12.2
  * kicadStepUpMod 10.23.0
  * Assembly4 0.50.7
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-19:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/17776

closed 2024-11-19 by maxwxyz

HoWilgh, 2024-11-19:
Closing it as 'not planned' is a bit strange. I would understand it, if it is because it's is a duplicate. 

Still, it should serve as a example to test, especially because it is quite simple to understand and test.