FreeCAD/FreeCAD #17968: isSame returns wrong result

RuiningLi, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To my understanding `isSame` compares the geometric properties of 2 entities, not for equality check for the identity of the object in memory.

However, consider the following code snippet
```
line = Part.LineSegment(App.Vector(0, 0, 0), App.Vector(0, 0, 1))
line_shape = line.toShape()
line_shape2 = line.toShape()
print(line_shape.isSame(line_shape2)  # False is printed
```

I think the behavior is not expected.

I'm very new to FreeCAD so might have some misunderstanding. But why is the above behavior like that?

### Full version info

```shell
freecad                   0.21.2          py311h94c7332_1    conda-forge
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-11-19:
Sorry, but your understanding is wrong. There are three related functions:
This is taken from the documentation of `TopoDS_Shape`.
* isPartner: Returns True if two shapes are partners, i.e. if they share the same TShape. Locations  and Orientations may differ.
* isSame: Returns True if two shapes are same, i.e. if they share the  same TShape with the same Locations. Orientations may differ.
* isEqual: Returns True if two shapes are equal, i.e. if they share the same TShape with the same Locations and Orientations.

Because you have created two independent shapes all of these functions will return False.

Here is an example to see how the functions can be used:
```
line = Part.LineSegment(App.Vector(0, 0, 0), App.Vector(0, 0, 1))
line_shape = line.toShape()

# The same object
line_shape2 = line_shape
line_shape.isSame(line_shape2)
# True, of course
line_shape.isPartner(line_shape2)
# True, of course
line_shape.isEqual(line_shape2)
# True, of course

# The reversed object
line_shape3 = line_shape.reversed()
line_shape.isSame(line_shape3)
# True because the location is still the same
line_shape.isPartner(line_shape3)
# True because the TShape is still the same
line_shape.isEqual(line_shape3)
# False because the orientation has changed

# Change the location of the reference shape
line_shape.applyTranslation(App.Vector(1,2,3))
line_shape.isSame(line_shape3)
# False because locations are different
line_shape.isPartner(line_shape3)
# True because the TShape is still the same
line_shape.isEqual(line_shape3)
# False because the orientation and location have changed
```

RuiningLi, 2024-11-19:
I see! Thank you for your clarification.

As a follow-up, Is there a function that can check equality for the geometry itself, regardless of the shape identities?

FlachyJoe, 2024-11-19:
You can check the boolean common result.
```python
(not (shape1.isNull() or shape2.isNull())) and shape1.common(shape2).isNull()
```

luzpaz, 2024-11-19:
@RuiningLi Please update to v1.0 stable, you're using v0.21.2

Please continue conversation on API usage on the forum.freecad.org

TIA

closed 2024-11-19 by luzpaz

wwmayer, 2024-11-19:
https://github.com/FreeCAD/FreeCAD/pull/17989