FreeCAD/FreeCAD #17971: FreeCAD crashes when creating BIM projects

phillipunzen, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

We are trying to map several halls in FreeCAD with the BIM module in version 1.0.0 RC4. As soon as 2 - 3 walls have been drawn, the program crashes without any feedback. We also tested whether the problem occurs on a different computer or with the RC2 version. Unfortunately, I have to say yes to both.

### Full version info

```shell
Version: 1.0.0RC4
Revision: 39104 (Git)
Releasedate: 2024/11/14
Operatingsystem: Windows 11 build 22631
Architecture: x86_64
Branch: (HEAD detached at 1.0rc4)
Hash: 8364675
```

### Subproject(s) affected?

BIM

### Anything else?

Lengths of more than 10 meters in each direction were specified. This "template" that is shown in the background with 1000mmx1000mm is exceeded accordingly. Could this be a cause? No other settings were made in the software.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-19:
Can you provide a demo file and more info about the steps you do?
@yorikvanhavre FYI

phillipunzen, 2024-11-19:
Unfortunately, I cannot create a demo file because it does not save.

Procedure:
1. Open FreeCAD
2. Click on "Create BIM Project"
3. Draw wall 1 (length 20m)
4. Draw wall 2 (length 50m)
5. Change the length of wall 2 afterwards
6. FreeCAD crashes

What I was able to determine, however, is that I performed the same procedure on a computer without a dedicated GPU, and it seemed to work more stable here. An NVIDIA P1000 is installed in each of the two "crashing computers". Could this be a cause?

luzpaz, 2024-11-20:
@phillipunzen can you update to 1.0.0 or weekly, test and report back. Please also include full [About](https://wiki.freecad.org/About) info when you do. TIA

@semhustej can you reproduce ?

semhustej, 2024-11-20:
Hi,

I am not able to reproduce with the 1.0 release. There was an issue with FC randomly crashing after multiple walls were created in RC2, but it has been fixed in RC3.

Few questions for @phillipunzen:

- by 'Create BIM Project' you mean using the BIM_Project tool or BIM_ProjectManager tool? (The name of the command is in the tooltip when hovering over the icon). 
- Can you please test if the crashes happen without this step in version 1.0?
- are you changing the length of the wall by changing the value of the 'Length' parameter in properties of the wall?

`This "template" that is shown in the background with 1000mmx1000mm is exceeded accordingly. Could this be a cause?` - This should not be the issue.

FEA-eng, 2025-09-14:
I don't have a crash. I only got a freeze when drawing the first wall.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

semhustej, 2025-09-14:
> > Can you please expand on what `freeze` means?
> 
> That FreeCAD was unresponsive (hung and processing) for a while.

Thanks. I get these 'freezes' (FreeCAD was unresponsive (hung and processing) for a while (sometimes 1 second, sometimes 5 seconds) ) with BIM and Draft objects (it can be wall or line object for instance). It happens only once after the WB is loaded. I am not able to pinpoint when it is more pronounced, so I did not report it. It already happens a long time in 1.1 development timeline. So I can confirm this, but it's not related to the OP issue. I don't know if it is worth reporting as it's very un-specific behavior.

semhustej, 2025-09-14:
As the original poster is not able to provide further feedback and others are not able to reproduce the OP issue, I think this issue should be closed.

closed 2025-09-26 by Roy-043