FreeCAD/FreeCAD #17973: The View > Toolbars menu shows toolbars without names. The user cannot know what to turn on/off!

raindropsfromsky, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The **View > Toolbars** menu shows toolbars without names. 
[image]

The user cannot know what to turn on/off!

**Desired:** 
Label these toolbars.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Beltrami 1.1.0
  * btl 0.9.9
  * CurvedShapes 1.0.12
  * Curves 0.6.51
  * ExplodedAssembly
  * fasteners 0.5.29
  * FreeCAD-Ribbon 1.3.17.2
  * OpenTheme 2024.9.1
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * SearchBar 1.0.2
  * sheetmetal 0.5.3
  * Silk 0.1.5
  * TabBar
  * Trails 2022.1.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-19:
This seems a duplicate of #16832.

luzpaz, 2024-11-19:
Closing in favor of #16832