FreeCAD/FreeCAD #17976: Color/Shading Of Model Changes After Editing Sketch

LHC-Canada, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have tested this with both dev version 0.22.39176 and 0.22.39240 with the same result.

Editing a sketch slightly, results in the graphics changing color/shading on the screen, but if the model is saved, exiting from FreeCad and relaunching the program and re-loading the model file, the graphics are  correct again.  It seems to be a non persistent issue caused by the recalculation process.

I have attached an example file and a couple of pictures to show the issue.

Steps to reproduce - 
- Open FreeCad and load the Test file
- Alter the center hole in sketch009 slightly – ie increase the diameter by 1mm from 5mm to 6mm
- Close the sketch and view the model

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/ProDark/Qt default
Installed mods: 
  * A2plus 0.4.56a
  * Beltrami 1.1.0
  * Curves 0.4.4
  * Defeaturing
  * ExplodedAssembly
  * fasteners 0.3.44
  * fcgear 1.0.0
  * IconThemes
  * kicadStepUpMod 10.13.0
  * Manipulator 1.4.5
  * QuickMeasure 2022.10.28
  * scripts
  * sheetmetal 0.4.24
  * VendorParts
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]
[Test.zip](https://github.com/user-attachments/files/17817305/Test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-19:
Confirm

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: lattice2 1.0.0 CurvedShapes 1.0.13 MeshRemodel 1.10.34 freecad.gears 1.3.0 fasteners 0.5.31 sheetmetal 0.5.6 PieMenu 1.9.5 (Disabled) Curves 0.6.51

```

maczargit, 2024-11-19:
To workaround (get back shading) this issue, you need to set DrawMode to FlatLines and click "Random Color" on Parts right-click menu in Model tree-list.
Can use it instead relaunching Freecad.

closed 2024-11-19 by maxwxyz