FreeCAD/FreeCAD #17977: Path not following Taper

KR-Huck, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Given a pad with a Taper Angle on the sides. Adaptive Profiling ignores the taper and insists on clearing the inside. Using Debian Linux Freecad 0.22.0dev  also windows Freecad 0.21.2 Regular Profile also ignores the Taper Angle. 

### Full version info

```shell
OS: Pop!_OS 22.04 LTS (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git) AppImage
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-19:
@KR-Huck please test on latest stable https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0.0 

If you can reproduce, please let us know.

KR-Huck, 2024-11-19:
I tried it with the 1.0 release and the outcome is the same.

luzpaz, 2024-11-20:
@KR-Huck please add full [About](https://wiki.freecad.org/About) info of 1.0.0 you tested on.

KR-Huck, 2024-11-20:
[test.zip](https://github.com/user-attachments/files/17833266/test.zip)

OS: Pop!_OS 22.04 LTS (LXDE/LXDE/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

KR-Huck, 2024-11-24:
Has this issue died?

luzpaz, 2024-11-24:
No, just pending. We don't have enough CAM devs available ATM. Perhaps rallying support on the forum ?

KR-Huck, 2024-11-24:
Thanks, I'm trying to figure out the Forum and ask questions. Seems I'm having trouble logging in. Surely something I'm doing incorrectly!

luzpaz, 2024-11-24:
@KR-Huck a forum moderator needs to approve your account before you can post. You can also try the discord chat https://discord.gg/w2cTKGzccC

KR-Huck, 2024-11-24:
The Forum worked.

luzpaz, 2024-11-25:
@KR-Huck can you kindly link the forum thread to this ticket?

KR-Huck, 2024-11-26:
OS: Pop!_OS 22.04 LTS (LXDE/LXDE/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

Not sure how to link, I can explain what I've learned form the forum that worked! See FreeCad help Cam Workbench - 3D Operations WATERLINE. The 3rd party OpenCamLib must be installed and set Preferences - Enable OCL dependent features. This is explained in the help.
3D Operations Waterline tool is now available! 
[image]

KR-Huck, 2024-11-26:
Solution provided by Legros

KR-Huck, 2024-11-26:
Thanks to all that helped!

closed 2024-11-26 by KR-Huck

luzpaz, 2024-11-26:
So essentially opencamlib dependency was missing and a preference needed to be enabled ?

KR-Huck, 2024-11-26:
This is true. The first time I tried this it didn't work for me and I thought this wasn't the solution. This was before asking on the Forum! Next time I tried (with tinkering) it worked. It is Experimental. I can't take any credit, my son actually fount the 3D pocket, Surface, Waterline and loaded the library on Linux.

luzpaz, 2024-11-26:
@KR-Huck the appimage you're using, where did you get it from ?

KR-Huck, 2024-11-26:
Print screen of my Linux machine

luzpaz, 2024-11-27:
@KR-Huck thanks for your replies. A couple more questions:

> This is explained in the help.

What help documentation are you referring to?

> 3rd party OpenCamLib must be installed

This is a given since OpenCamLib is installed in the AppImage AFIACT
https://github.com/FreeCAD/FreeCAD-Bundle/blob/3d88909069a7d23df9d2a3fd0dd9e5995037c1ea/conda/linux/create_bundle.sh#L30

KR-Huck, 2024-11-27:
The answer about Release was provided by my son the "Linux expert". Yes the OpenCamLib was included with the app image. The Help was from FreeCad User Documentation. Searching documentation for OpenCamLib I was convinced Windows installs includes this workbench and Linux required an install. This was not true, Linux has the OpenCamLib installed with the image. I did read this so possibly at some point Linux did require more work? 

Simply, use the latest stable version and set Preferences - Enable OCL dependent features. These are some fantastic features!