FreeCAD/FreeCAD #17978: Draft Snap point color with Light Theme

semhustej, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the default Light Theme, Draft Snap point are "highlighted" in black color. It is possible to work with this setting, but it does not contrast enough compared to the default black geometry of lines. 
Additionally it can be also mistaken to the end points on lines, which are not sapped to.

See screenshot:
[image]

Proposed solution:
The snap color should be changed in Light Theme

As an example, this is snap color in BricsCAD with white background.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-11-25:
The snap point color is a preference value ( BaseApp/Preferences/Mod/Draft/snapcolor ). I think it can easily be themed ( @obelisk79 ). But maybe we could give it a color that works well on both light and dark background such as the orange you propose?

semhustej, 2024-11-25:
I tried saturated bright orange color #ffaa00 for both light and dark themes and it seems to work for me:

Light theme:

[image]
[image]

Dark theme:

[image]

[image]

The light theme might also depend on whether the default preselect hover color will be changed, which is this issue: https://github.com/FreeCAD/FreeCAD/issues/17916

obelisk79, 2024-11-26:
There are a few (big in my opinion) issues with selection color. However, yes, the selection color is themable/customizable. There just isn't a distinction between 2d and 3d geometry which causes problems with lighting/shading.

semhustej, 2024-11-26:
> There are a few (big in my opinion) issues with selection color. However, yes, the selection color is themable/customizable. There just isn't a distinction between 2d and 3d geometry which causes problems with lighting/shading.

Yes, this is true. I understand the visibility for 3D mechanical modelling is a priority.

I think in the case of orange (or other than black) snap points color works well enough for all purposes and could be the default ones (especially for the Light Theme). 

With regards to selection: it is good that colors are customizable, so each user can set their preferences. The goal should be for the defaults should be at least good enough for all applications.

Roy-043, 2024-11-26:
Currently each theme (Light, Dark and Classic) have their own value for this color. Just changing this in the Draft preferences would have no effect. Some coordination is needed.

PR #18949 (2025-01-09): Draft (and Pref Packs): fix snapcolor
Fixes #17978.
commit f6f11ac4 (2025-01-09): Draft (and Pref Packs): fix snapcolor

Fixes #17978.
closed 2025-01-09 by Roy-043