FreeCAD/FreeCAD #17979: Dark Theme: 2D geometry with little contrast to the background color

semhustej, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is too little contrast of 2D geometry with default color to the background color when using the default Dark Theme. It is not critical, it is possible to work in 2D with the default setup, but not too pleasant.

See screenshot of a 2D drawing:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Core

### Anything else?

I am not sure if there is a solution that would not break other things, especially rendering of 3D objects.

In standard 2D drafting software, the default black lines are usually rendered white if user chooses a black background.

I know it is possible to adjust the background color, but a new user opening FreeCAD for the first time needs to ne taken into account, this issue is about the defaults.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-11-19:
Maybe it is possible to follow the approach for Sketcher when @kadet1090 fixed the problem for that workbench.

maxwxyz, 2024-11-20:
@MisterMakerNL FYI

MisterMakerNL, 2025-08-30:
Reading the forum and the last post I think this can only be fixed by giving draftsight object their own sketcher like color system. Which I think was @kadet1090 opinion too.

R2V1W3, 2026-09-14:
Draft lines need their own color setting parameter back, removing draft dedicated visual parameters was wrong decision IMO. 
I am sure people will settle even if the line color default uses draft annotation color parameter, or the sketcher line color if adding new parameters is a problem. Managing color through layers isn't working because many tools create draft objects before user has a chance to create suitable colored layer. 
The line color also applies to line color during creation of object, dragging line or polyline, as an example.  

Drafts are among the most important objects in many workflows.

Wireframe also needs its own color parameter, but that should be another issue it doesn't have to be solved at once if it means Draft will be waiting for a solution indefinitely.

Roy-043, 2026-09-20:
@FreeCAD/design-working-group Maybe you can chip in here. For me this is primarily a theming issue. One way to solve it would be to change the background color in the Dark theme as Max has suggested (but that may be controversial 😄 )

maxwxyz, 2026-09-20:
Workaround for now: https://github.com/FreeCAD/FreeCAD/pull/32827

closed 2026-09-22 by kadet1090

semhustej, 2026-10-01:
Tested with the latest weekly. The situation is still not ideal, but better than previous state. Thank you.

```
OS: Debian GNU/Linux 13 (trixie) (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 26.3.0dev.20261001 (Git shallow) AppImage
Build date: 2026/10/01 12:05:00
Build type: Release
Branch: (HEAD detached at 99c5620c5)
Hash: 99c5620c5f739ca5c308849dda95ac468fec8b79
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Revit/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/158.196/1
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 25.0.7-2+deb13u1/Intel/Mesa Intel(R) HD Graphics 4600 (HSW GT2)

```