FreeCAD/FreeCAD #17980: still wrong feed speeds

jeffgrain, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

as referenced in #17848 not everything got fixed also reference #17871 .. still generates the wrong feed speeds

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

included is screen capture , part file , and gcode file..
[test.zip](https://github.com/user-attachments/files/17818999/test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

atapi, 2024-11-19:
Try to change unit system from "Standard" to "Metric small parts _CNC"
[image]

jeffgrain, 2024-11-19:
Ok.. I did change that originally.. but it didn't save.. I have it changed now.. but that created a different problem with choosing the origin point ,, where it didn't in the first test.. a little busy today.. but I need to do some testing and get some example files to be 110 percent sure..  but using the small parts did fix the feed speed..

closed 2024-11-20 by maxwxyz