FreeCAD/FreeCAD #17981: Detail View tool in Techdraw

Braulio-B, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi! I'm a new Freecad user. I'd like to report a bug to help the development team.

The Detail View tool is not working correctly. I cannot change the circle linestyle of the detail highlight.

Thank you for all the amazing work you are doing!

[image]

### Full version info

```shell
I tried to change it in Edit>> Preferences>> TechDraw>> Annotation>> Datail Highlight Style, and after restart Freecad it didn't work too.

I'm not a programer, I can only help reporting. I hope it help you.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-20:
Thanks. Please edit the above ticket and place the full [About](https://wiki.freecad.org/About) info into the appropriate section of the template. TIA

WandererFan, 2024-11-20:
You need to change the highlight style on the base view (ie the view on which the highlight will be drawn), not the detail view.

[image]

closed 2024-12-11 by luzpaz