FreeCAD/FreeCAD #17982: Freecad 1.0rc4: Display errors and main window resizing

InRiPa, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Top down menus are not displayed although it seems they are there, just "hidden". 
Rearranging the main window is not possible at the moment. Only Maximized works. 
I made a video displaying the issues one by one (having these on two different thinkpads):
1.  Top menu with `file`, `edit` works without issues, all the menus below it are affected.
2. 00:32 - Clicking on the small top down arrow doesn't show anything. However, hovering under it i can see the tooltips. So they are somehow there. I clicked multiple times, so this is a constant behavior (contrary to mouse navigation later)
3. 00:46 - Same for the other drop down menus.
4. 00:57 - Also within a Sketch
5. 01:11 - The mouse navigation style sometimes opens the menu sometimes i doesn't.  
6. 01:30 - Same thing with with isometric view
7. 01:40 - Displays the issue with all kinds of resizes

https://github.com/user-attachments/assets/c78d6a88-467a-4fa1-9670-f1c98152edcb

### Full version info

```shell
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git)
Build type: Release
Branch: makepkg
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/fusion
Installed mods: 
  * fasteners 0.4.67
  * lattice2 1.0.0
  * dodo 1.0.0
  * A2plus 0.4.60n
  * Assembly4 0.50.4
  * Curves 0.6.13
```

### Subproject(s) affected?

None

### Anything else?

See added video

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-19:
Can not confirm in Archlinux and KDE

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git)
Build type: Release
Branch: makepkg
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
Installed mods: lattice2 1.0.0 CurvedShapes 1.0.13 MeshRemodel 1.10.34 freecad.gears 1.3.0 fasteners 0.5.31 sheetmetal 0.5.6 PieMenu 1.9.5 (Disabled) Curves 0.6.51
```

InRiPa, 2024-11-20:
Hey,
thanks for helping! Ye since i couldn't find similar issues, i was already under the impression that this is only me. I'm just not sure how to figure out the problem. 

I tried the `safe mode` but the issues are the same. 
I wonder if the `Coin warning in glxglue_isdirect(): Couldn't get current GLX context.` is related? 

from a LLM:
> GLX: This stands for OpenGL Extension to the X Window System. It is a protocol that allows OpenGL to interface with the X   Window System, which is commonly used on Unix-like operating systems.

Even though I am on Wayland, maybe related to XWayland? 

Currently I'm building the latest version from git (using AUR) - if the build succeeds, I'll try this one. 
Otherwise I'll try with just one Monitor. I noticed, that some applications have issues with multi monitor setups. 

--
Oh, I think I made some progress on nailing it down. 
Through the safe mode, the normal `freecad` didn't start maximized. So I could try the "non-maximized" version. 

1. The middle menus behaved now, like the bottom mouse navigation selection. They partially appeared here and there. The issue is that i hover over the drop-down and **must** wait until the yellow tool-tip appears. Then the drop-down will work. If i quickly go over it and click on it, it never pops up. 
2. Resizing the window by dragging the corner, it seems that only certain sizes work. If i drag slowly, I can see that certain sizes are distorted and some bigger (further dragging) make it work again.  It feels like, after resizing not everything gets updated in time.

In the journal log, the following entry appeared:
> Nov 20 08:56:07 <laptop> gnome-shell[1544]: Buggy client caused popup to be placed outside of parent window

tarman3, 2024-11-20:
I tryed run with commands
`QT_QPA_PLATFORM=wayland freecad`
and
`QT_QPA_PLATFORM=xcb freecad`
No issue like you show.

Maybe you can try to boot in some Linux live session. E.g. Manjaro, Ubuntu, ...

InRiPa, 2024-11-20:
I think the major difference between our setups is that you're using `KDE` and I'm using `GNOME`. One based on `Qt`, the other on `Gtk`. Since it seems to be a graphical issue with window drawing, I think the difference matters. Probably this is an issue with `GNOME`/`Gtk` on my side. 

I will check the version when my mirror got it :) (still shows extra/freecad 1.0rc4-3). the git build is also still compiling.

nmschulte, 2024-11-23:
@InRiPa how many displays do you have?

InRiPa, 2024-11-24:
@nmschulte 
A setup with one additional monitor and a setup with two additional monitors. 
But I don't think this is related anymore. I tried it without any monitor connected and I have these issues as well.

In regards to the window resizing issues. I noticed that when just opening freecad, I can drag the window as i want. All is always fine. The resizing issues of the main window occur the moment i have a project/sketch open.
Also, the moment I open a project, I get `Coin warning in glxglue_isdirect(): Couldn't get current GLX context.`.

Probably unrelated. I tried to find a workbench with a drop down menu, when no project is loaded. On assembly4 I get 
`/home/user/.local/share/FreeCAD/Mod/Assembly4/./configurationEngine.py:607: SyntaxWarning: invalid escape sequence '\;'
  badChars = '`~!@#$%^&*()-+=|\;:\'".,'
Initializing Assembly4 workbench (0.50.15) ............................. done.
`
The `sync view` has a drop down menu. Same issue when no project is loaded. Nothing appears.
However, it only appears when I have the main window not maximized. Additionally, in non maximized mode, I wanted to drag it over to the second monitor. It immediately resized itself and locked this size. I can't resize it anymore in X-direction. only in Y-direction. (Couldn't repeat that on restart)

What really boggles me is that I seem to be the only one. And that even on two devices with a quite common setup. 

### Trying AppImage

I downloaded the **AppImage** and in there everything **seems to work**.
However, the journal goes crazy during the usage. 

The first thing, it seems to launch `X11`?
Then there seems to be a lot of errors related related to the window and it keeps coming (but not flooding).

```
Nov 24 11:15:40 HostPC gnome-shell[1554]: Could not issue 'GetUnit' systemd call
Nov 24 11:15:40 HostPC systemd[1447]: Reached target GNOME session X11 services.
Nov 24 11:15:40 HostPC systemd[1447]: Starting GNOME XSettings service...
Nov 24 11:15:40 HostPC gnome-shell[1554]: Failed to launch ibus-daemon: Failed to execute child process “ibus-daemon” (No such file or directory)
Nov 24 11:15:40 HostPC gnome-shell[3800]: The XKEYBOARD keymap compiler (xkbcomp) reports:
Nov 24 11:15:40 HostPC gnome-shell[3800]: > Warning:          Unsupported maximum keycode 708, clipping.
Nov 24 11:15:40 HostPC gnome-shell[3800]: >                   X11 cannot support keycodes above 255.
Nov 24 11:15:40 HostPC gnome-shell[3800]: > Warning:          Could not resolve keysym XF86KbdInputAssistPrevgrou
Nov 24 11:15:40 HostPC gnome-shell[3800]: > Warning:          Could not resolve keysym XF86KbdInputAssistNextgrou
Nov 24 11:15:40 HostPC gnome-shell[3800]: Errors from xkbcomp are not fatal to the X server
Nov 24 11:15:41 HostPC gsd-xsettings[3798]: Failed to get current UI scaling factor: GDBus.Error:org.freedesktop.DBus.Error.NameHasNoOwner: Destination does not exist
Nov 24 11:15:41 HostPC systemd[1447]: Started GNOME XSettings service.
Nov 24 11:15:41 HostPC systemd[1447]: Reached target GNOME session X11 services.
Nov 24 11:15:41 HostPC mutter-x11-fram[3819]: Using GtkSettings:gtk-application-prefer-dark-theme with libadwaita is unsupported. Please use AdwStyleManager:color-scheme instead.
Nov 24 11:15:41 HostPC gnome-shell[1554]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x60000e specified for 0x600019 (FreeCAD).
Nov 24 11:16:07 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:08 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:09 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:10 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:11 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:11 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:13 HostPC gnome-shell[1554]: Window manager warning: WM_TRANSIENT_FOR window 0x60016d (FreeCAD) for 0x600173 (FreeCAD) window override-redirect is an override-redirect window and this is not correct according to the standard, so we'll fallback to the first non-override-redirect window 0x600110.
Nov 24 11:16:13 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:13 HostPC gnome-shell[1554]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x60000e specified for 0x60017b (FreeCAD).
Nov 24 11:16:14 HostPC gnome-shell[1554]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x60000e specified for 0x600183 (FreeCAD).
Nov 24 11:16:18 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:20 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:21 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:23 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:25 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:16:27 HostPC gnome-shell[1554]: meta_window_stick: assertion '!window->override_redirect' failed
Nov 24 11:17:22 HostPC systemd[1]: tmp-.mount_FreeCAHLnGcl.mount: Deactivated successfully.
```

nordtechtiger, 2024-12-26:
The invisible drop-down menus have started appearing for me too. Everything worked perfectly until I booted it up today - albeit I haven't been using FreeCAD for the last couple of weeks. I'm running the packaged version on Arch, and Gnome 47 under Wayland. Hovering over the buttons and allowing the tooltip to show before clicking makes the menu appear.
```
FreeCAD 1.0.0 Revision: 39109 (Git)
```

pzauner, 2025-01-28:
I have the same problem, using arch, GNOME and Wayland. The menu entries are visible when using either GNOME via Xorg or KDE via Wayland.

nmschulte, 2025-01-28:
Do you all launch FreeCAD w/ `QT_QPA_PLATFORM=xcb` under Wayland?  I am able to effectively use FreeCAD without it (it forces XWayland use), but must set `COIN_GL_NO_CURRENT_CONTEXT_CHECK=1`, so I tend to use the QT env var.

nordtechtiger, 2025-01-28:
I have since switched to running a v1.1 devbuild from the AppImage and all of my issues have been resolved. Not able to test whether this is because of the newer version or because of the different packaging (AppImage vs pacman).

InRiPa, 2025-09-02:
still an issue with for example version FreeCAD 1.0.2, Libs: 1.0.2R39319 (Git)

johnsonm, 2026-04-03:
I reproduce the hidden menu problem on a from-source build of 1.1, 44786.

@InRiPa are you testing 1.2 dev or 1.1 final?