FreeCAD/FreeCAD #17983: BIM dialogues undesirable behavior

semhustej, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Multiple BIM Workbench dialogue boxes behave undesirably. See video with example of Layers Manager.

1) When Layers Manager is opened, second instance of Layers Manager can be opened by clicking on the tool icon
Desired behavior: If Layers Manager is already open and user clicks on the icon, the open Layer Manager should come into focus

2) When Layers Manager is opened and user clicks outside of it into 3D view, the Layers Manage window is placed behind the main FreeCAD window and user does not see it.
Desired behvior: even if the Layers Manager window is non-modal (because of it's functionality) it should be set to always stay on top.

See both behaviors in the video:

https://github.com/user-attachments/assets/a091cf51-3642-4c7a-8a47-4e5fba198e6e

I observed the same issues with these BIM tools, I think all of them should behave as proposed above. Maybe some can be set to be modal, I am not sure as I do not use all the tools:

- BIM_Setup
- BIM_ProjectManager
- BIM_IfcElements
- BIM_IfcQuantities
- BIM_IfcProperties
- BIM_Classification
- BIM_Layers
- BIM_Material (only partly)
- maybe there are other ones....

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

I think Arch_Schedule tool has the behavior closest to ideal one. It would be nice if all above mentioned tools had the same behavior.

The two issues with this one, I am not sure if they are solvable:
- The window stays on top of FreeCAD window - that is desirable. But it also stays on top of windows of other programs (if for instance user switches to a web browser)
- When the tool window is open and user clicks the tool icon again, the location and size of the tool window is reset.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-11-25:
Ok these make sense indeed. I'll look into it

semhustej, 2025-06-28:
> > 2. When Layers Manager is opened and user clicks outside of it into 3D view, the Layers Manage window is placed behind the main FreeCAD window and user does not see it.
> 
> This does not happen on Ubuntu. 

I can confirm that the dialog behavior is correct on Debian + GNOME.
Not on Windows though.

semhustej, 2025-07-12:
> The problem is, if the layer manager is marked to stay on top, it stays on top of all other windows, which might not be what we want.

It should be possible for the dialogue to behave the correct way even on windows, as an example here is the correct behavior of the Material selection dialogue:

https://github.com/user-attachments/assets/63095125-d396-48a8-9521-905919ddbf60

semhustej, 2025-11-11:
> Is this exclusive to Windows ?

The issue does not occur on Linux, but it seems nobody checked the behavior on Mac.

paullee0, 2025-12-14:
New case found and being discussed on FC Forum :- 

- https://forum.freecad.org/viewtopic.php?p=862167#p862167
- https://forum.freecad.org/viewtopic.php?p=862131#p862084