FreeCAD/FreeCAD #17984: Spreadsheet: Black text shown white in freecad 1.0.0

reald, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Download Freecad 1.0.0 Appimage (x86_64)
2. Run Appimage (on ubuntu 24.04 with xfce)
3. On first start choose light theme
4. Create new file, create spreadsheet
5. Type some text into a spreadsheet cell

Result: Text color is white on white background

Other text foreground colors can be used normally but black is always shown white.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

huguesdpdn-aerospace, 2024-11-19:
Hello @reald ,

## Test

I test it on :

```
OS: Ubuntu 24.04.1 LTS (LXQt/Lubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39215 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b3a2268289482c65c18bbb4cc073b33020e71ce2
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```
so same OS but not on XFCE4 ... and after reseting all FreeCAD configuration and selecting light theme , I have white background and black test.

## Video

https://github.com/user-attachments/assets/b1443979-8047-4088-8c15-8841b71c5878

## HOWEVER

The color button to change the text color do **NOT** work, but the one for the background does. Interesting ...

## Requesting additional test
Can you, @reald , execute command:
`rm -rf ~/.local/share/FreeCAD/ ~/.config/FreeCAD/ ~/.cache/FreeCAD/`
and then run FreeCAD and see if the error persists ?

Waiting reporter feedback

hyarion, 2024-11-20:
(FYI safe mode: https://wiki.freecad.org/Std_RestartInSafeMode can also be used for temporarily testing without any user configurations)

semhustej, 2024-11-20:
I can confirm the behavior @huguesdpdn-aerospace is describing

- default text color is black
- I can change the cell background color
- when setting different text color, the text stays black
[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

luzpaz, 2024-11-20:
> default text color is black
> I can change the cell background color
> when setting different text color, the text stays black

I can confirm @semhustej's observation that changing font color doesn't register in the Spreadsheet. 

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

BUT 

> Result: Text color is white on white background

I can't reproduce OPs findings

reald, 2024-11-20:
Tried the supposed things in safe mode (./FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage --safe-mode);
- Original issue is NOT reproduceable in safe mode: Text color is black (if cell is not highlighted)
- Text color can be changed, correct color is shown if cell is not highlighted.

[image]

semhustej, 2024-11-20:
When I saw @reald screenshot, a thing popped into my mind and I tried this:

- Set Classic Theme UI style -> Spreadsheet text color changes according to cell setting
- Set Light Theme UI style -> Spreadsheet text color is always white, irrespective of cell setting
- Set Dark Theme UI style -> Spreadsheet text color is always black, irrespective of cell setting
- Set back Classic Theme UI style -> Spreadsheet text color changes according to cell setting

I think this is a Stylesheet issue related to UI Theme settings.

I think OP's issue is also related to this as I remember in some situations when switching themes, the text in Light Theme stayed white on white background. I was not able to reproduce that behavior though.

MisterMakerNL, 2024-11-20:
See:
https://github.com/FreeCAD/FreeCAD/issues/16950

hyarion, 2024-11-20:
IIRC, spreadsheet has a bug which makes it impossible to clear the text and background override color.
It's described in this issue: https://github.com/FreeCAD/FreeCAD/issues/13862 (and it references a PoC clear color branch I made)

MisterMakerNL, 2024-11-20:
As stated the text color is bugged, there was no way to get this to work with stylesheets without bugs.
Opentheme uses a different method but if you open a spreadsheet made in opendark in open light you get white text on white cells. If OP has used opentheme this could be the issue here.
Also op Screenshots shows classic, the version info also shows classic.

reald, 2024-11-20:
Indeed I confused classic with light theme. Using light theme I have the same issue described above, font color cannot be changed.

huguesdpdn-aerospace, 2024-11-21:
I was working on my project and then, pouf, @reald 's problem appears when I just created a Spreadsheet with light theme:
[image]
[image]

So YES, I did had white text on white cells, but I don't know how to reproduce it.
It seems we have a double bug :/

kwoqifeli22, 2024-12-11:
I can confirm a similar behavior with FreeCAD 1.0 using the Classic Theme on Windows 10 (12th Gen Intel i9-12950HX, NVIDIA RTX A3000). Adjusting the text color to RGB(0, 0, 0) causes the text to disappear unless the cell is selected. Switching to other themes results in slightly different behavior, as described in this issue.

Roy-043, 2025-01-25:
The problem is that these settings are not included in "FreeCAD Classic.cfg" and not exposed in the Preferences. You can find (and edit or remove) them with [Std_DlgParameter](https://wiki.freecad.org/Std_DlgParameter).
BaseApp > Preferences > Mod > Spreadsheet

To avoid such issues the preference packs should all include the exact same preferences.

closed 2025-01-27 by Roy-043

rapax00, 2025-06-30:
> The problem is that these settings are not included in "FreeCAD Classic.cfg" and not exposed in the Preferences. You can find (and edit or remove) them with [...]

You're genius! Thanks

PR #19245 (2025-01-27): PrefPacks: Add spreadsheet colors to classic theme
Fixes #17984.
commit bad2b8b0 (2025-01-27): PrefPacks: Add spreadsheet colors to classic theme

Fixes #17984.
commit 0e3d2a20 (2025-02-16): PrefPacks: Add spreadsheet colors to classic theme

Fixes #17984.