FreeCAD/FreeCAD #17985: Addon Manager wants to update to lower or same version

numo68, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Addon manager wants to update to a lower or even same version of an addon. I did not dare to actually try; perhaps this is just a display problem, perhaps other fields than just the version are used in the comparison, perhaps something completely different.

[image]

Thanks

### Full version info

```shell
OS: macOS 15.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * DynamicData 2.60.0
  * freecad.gears 1.0.0
  * Defeaturing 1.2.2
  * A2plus 0.4.64
  * fasteners 0.5.14
  * lattice2 1.0.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-19:
Have the same issue with SheetMetal Workbench.
Refresh local cache resolved issue.

numo68, 2024-11-19:
yep, seems to work here as well.