FreeCAD/FreeCAD #17987: Fails to build with libc++ 19.1.x

MehdiChinoune, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Fails to build with libc++ 19.1.x
```
  FAILED: src/Base/CMakeFiles/FreeCADBase.dir/__/CXX/cxxsupport.cxx.obj 
  D:\M\msys64\clang64\bin\clang++.exe -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_CONTAINER_DYN_LINK -DBOOST_CONTAINER_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK=1 -DBOOST_FILESYSTEM_NO_LIB -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DBOOST_THREAD_USE_DLL -DFCBase -DFMT_SHARED -DFreeCADBase_EXPORTS -DHAVE_CONFIG_H -DHAVE_D3D -DHAVE_FFMPEG -DHAVE_FREEIMAGE -DHAVE_FREETYPE -DHAVE_OPENGL_EXT -DHAVE_OPENVR -DHAVE_RAPIDJSON -DHAVE_SNPRINTF -DHAVE_SWIG=1 -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DOCC_CONVERT_SIGNALS -DPYCXX_6_2_COMPATIBILITY -DPYCXX_DLL -DQT_CORE_LIB -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DUNICODE -DVTK_OPENGL2_BACKEND -DWIN32 -DWIN64 -DWINVER=0x0A00 -DZIPIOS_UTF8 -D_ENABLE_EXTENDED_ALIGNED_STORAGE -D_OCC64 -D_UNICODE -D_WIN32_WINNT=0x0A00 -D_WIN64 -IC:/_/B/src/build-CLANG64/src/Base/FreeCADBase_autogen/include -IC:/_/B/src/build-CLANG64 -IC:/_/B/src/build-CLANG64/src -IC:/_/B/src/FreeCAD-1.0.0/src -IC:/_/B/src/build-CLANG64/src/Base -IC:/_/B/src/FreeCAD-1.0.0/src/Base -ID:/M/msys64/clang64/include/python3.12 -isystem D:/M/msys64/clang64/include/qt6/QtCore -isystem D:/M/msys64/clang64/include/qt6 -isystem D:/M/msys64/clang64/share/qt6/mkspecs/win32-clang-g++ -Wall -Wextra -Wpedantic -Wno-write-strings -march=nocona -msahf -mtune=generic -O2 -pipe -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wp,-D__USE_MINGW_ANSI_STDIO=1 -Wno-macro-redefined -Wno-deprecated-declarations -Wno-unknown-pragmas -Wno-attributes -O3 -DNDEBUG -std=gnu++20 -pthread -MD -MT src/Base/CMakeFiles/FreeCADBase.dir/__/CXX/cxxsupport.cxx.obj -MF src\Base\CMakeFiles\FreeCADBase.dir\__\CXX\cxxsupport.cxx.obj.d -o src/Base/CMakeFiles/FreeCADBase.dir/__/CXX/cxxsupport.cxx.obj -c C:/_/B/src/FreeCAD-1.0.0/src/CXX/cxxsupport.cxx
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/cxxsupport.cxx:42:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/cxxsupport.cxx:38:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Objects.hxx:42:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Objects.hxx:45:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Exception.hxx:47:
  D:/M/msys64/clang64/include/c++/v1/string:820:42: error: implicit instantiation of undefined template 'std::char_traits<unsigned int>'
    820 |   static_assert(is_same<_CharT, typename traits_type::char_type>::value,
        |                                          ^
  C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Objects.hxx:2178:28: note: in instantiation of template class 'std::basic_string<unsigned int>' requested here
   2178 |             return *this = *rhs;
        |                            ^
  D:/M/msys64/clang64/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
     23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
        |                             ^
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/cxxsupport.cxx:42:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/cxxsupport.cxx:38:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Objects.hxx:42:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Objects.hxx:45:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Exception.hxx:47:
  In file included from D:/M/msys64/clang64/include/c++/v1/string:647:
  D:/M/msys64/clang64/include/c++/v1/string_view:300:42: error: implicit instantiation of undefined template 'std::char_traits<unsigned int>'
    300 |   static_assert(is_same<_CharT, typename traits_type::char_type>::value,
        |                                          ^
  D:/M/msys64/clang64/include/c++/v1/__type_traits/is_convertible.h:22:99: note: in instantiation of template class 'std::basic_string_view<unsigned int>' requested here
     22 | struct _LIBCPP_TEMPLATE_VIS is_convertible : public integral_constant<bool, __is_convertible(_T1, _T2)> {};
        |                                                                                                   ^
  D:/M/msys64/clang64/include/c++/v1/string:745:29: note: in instantiation of template class 'std::is_convertible<unsigned int *const &, std::basic_string_view<unsigned int>>' requested here
    745 |     : public _BoolConstant< is_convertible<const _Tp&, basic_string_view<_CharT, _Traits> >::value &&
        |                             ^
  D:/M/msys64/clang64/include/c++/v1/string:1161:27: note: in instantiation of template class 'std::__can_be_converted_to_string_view<unsigned int, std::char_traits<unsigned int>, unsigned int *>' requested here
   1161 |             __enable_if_t<__can_be_converted_to_string_view<_CharT, _Traits, _Tp>::value &&
        |                           ^
  D:/M/msys64/clang64/include/c++/v1/string:1165:42: note: while substituting prior template arguments into non-type template parameter [with _Tp = Py_UCS4 *]
   1165 |   _LIBCPP_CONSTEXPR_SINCE_CXX20 explicit basic_string(const _Tp& __t, const allocator_type& __a)
        |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   1166 |       : __r_(__default_init_tag(), __a) {
        |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   1167 |     __self_view __sv = __t;
        |     ~~~~~~~~~~~~~~~~~~~~~~~
   1168 |     __init(__sv.data(), __sv.size());
        |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   1169 |   }
        |   ~
  C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Objects.hxx:2240:24: note: while substituting deduced template arguments into function template 'basic_string' [with _Tp = Py_UCS4 *, $1 = (no value)]
   2240 |             ucs4string ucs4( buf, size() );
        |                        ^
  D:/M/msys64/clang64/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
     23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
        |                             ^
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/cxxsupport.cxx:42:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/cxxsupport.cxx:38:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Objects.hxx:42:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Objects.hxx:45:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Exception.hxx:47:
  D:/M/msys64/clang64/include/c++/v1/string:2343:3: error: implicit instantiation of undefined template 'std::char_traits<unsigned int>'
   2343 |   traits_type::copy(std::__to_address(__p), __s, __sz);
        |   ^
  D:/M/msys64/clang64/include/c++/v1/string:1078:5: note: in instantiation of member function 'std::basic_string<unsigned int>::__init' requested here
   1078 |     __init(__s, __n);
        |     ^
  C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Objects.hxx:2240:24: note: in instantiation of member function 'std::basic_string<unsigned int>::basic_string' requested here
   2240 |             ucs4string ucs4( buf, size() );
        |                        ^
  D:/M/msys64/clang64/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
     23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
        |                             ^
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/cxxsupport.cxx:42:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/cxxsupport.cxx:38:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Objects.hxx:42:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Objects.hxx:45:
  In file included from C:/_/B/src/FreeCAD-1.0.0/src/CXX/Python3/Exception.hxx:47:
  D:/M/msys64/clang64/include/c++/v1/string:2344:3: error: implicit instantiation of undefined template 'std::char_traits<unsigned int>'
  
   2344 |   traits_type::assign(__p[__sz], value_type());
  
        |   ^
  
  D:/M/msys64/clang64/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
  
     23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
  
        |                             ^
  
  4 errors generated.
```

### Full version info

```shell
1.0.0
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hmartinez82, 2024-11-20:
Probably related: 4th paragraph at https://libcxx.llvm.org/ReleaseNotes/19.html#deprecations-and-removals

hmartinez82, 2024-11-20:
Is this an acceptable workaround? https://github.com/msys2/MINGW-packages/pull/22598/files#diff-8d212fb30c9973d8eb4fddd36770920f8e91bc0a3c978b4f583ec84b7159639c

luzpaz, 2024-11-22:
@chennes any windows package builders we can deffer this to ?