FreeCAD/FreeCAD #17991: CAM: Unnecessary tool changes when deactivating an array of operations

Huwyler, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating an array of of an operation (e.g. an array of a pocket), and then deactivating it (e.g. because you had to interrupt the CNC process and want to start again after that array operation), it keeps the tool change for that action, resulting in subsequent tool changes between the action before and after the deactivated array.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * OpticsWorkbench 1.0.24
  * Gridfinity 0.6.0
  * freecad.gears 1.2.0
  * fasteners 0.5.28
  * FeedsAndSpeeds 0.5.0
  * CfdOF 1.27.11
  * EM 2.1.1
  * sheetmetal 0.4.26
  * lattice2 1.0.0
  * Curves 0.6.50
  * dodo 1.0.1
  * btl 0.9.9
  * CurvedShapes 1.0.10
  * Assembly4 0.50.15
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-20:
Can you upload an example file and steps how to reproduce? Can you reproduce the issue in the 1.0 version?

Huwyler, 2024-11-21:
Oh, WOW, I haven't noticed that 1.0 is out! Great news!

Sorry, this is now quite embarrassing: I cannot reproduce it on FC1.0, and I even think, I had this issue with the link branch, not the official FC, as, when I wanted to reproduce it, I noticed, that the array operation is just a separate operation, with its base being the operation to copy - I remember it being shown more like a dress up.

But I noticed another minor bug: Deactivating an array works just fine (without unnecessary tool changes), but in the tree view it is not shown as such. 

I don't think it makes sense to add a file as it is very easy to reproduce: Create a simple thingy with a pocket, change to CNC, create a pocket operation and then create an array out of it. Then deactivate the array operation. It looks like it is just hidden, but in the properties one can see that it is deactivated, and also, it is removed from the GCode.

Shall I close this and open another one, or do we just rename this one accordingly?
Again, sorry for the confusion!

maxwxyz, 2024-11-21:
I'm closing this. You can check if there is already a similar issue in CAM and add your findings there. If there is no such issue, feel free to create one.

closed 2024-11-21 by maxwxyz