FreeCAD/FreeCAD #17992: After changing some parameters, the external geometry type changes from "Circle" to Ellipse", after which the external geometry breaks.

Mustlane, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The way to reproduce:
> Open Sketch003
> Try using an "External Geometry" tool on Body001.Edge21.
> Create a sketch like this
[image]
>Change the value of Sketch002.Edge10 (e.g. 6mm)

And then an external edge that you used in Sketch003 suddenly changes from the type "Circle" to the type "Ellipse" and then the whole Sketch002 just breaks like this:
[image]

------------------------------------------------------------------------------------------------------
While trying to reproduce the problem, I also saw another strange behavior (even tho it happened only twice):

Body001.Edge21 doesn't show in Sketch003 when trying to use "External Geometry".
Only the center point (since it is a part of an ellipse) of it appears, but not the edge itself, like this:
[image]

### Full version info

```shell
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Sketcher

### Anything else?

The file itself:
[Example.zip](https://github.com/user-attachments/files/17822472/Example.zip)

-----------------------------------------------------------------------------------
Also, it is kinda my first issue submit ever. Tell me if I forgot to add something or didn't explain the problem well enough.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-21:
I have trouble recreating the issue as the view is clipped in the active sketch as soon as I orbit.
@FlachyJoe @Rexbas any idea here?
[image]

FlachyJoe, 2024-11-21:
Confirmed.

@Mustlane 
> Tell me if I forgot to add something or didn't explain the problem well enough.

Very good first issue, thank you!

@maxwxyz I can not reproduce your clipping issue but I'm using my dev branch including the infinite sketch axes which can fix this.

luzpaz, 2026-05-05:
@drwho495 is this toponaming related ?