FreeCAD/FreeCAD #17995: Dimensions Incorrect when Importing to Blender

SlyCoop91, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is probably a glitch or something when creating a spur gear in FreeCAD.  It relates to the gear's dimensions after importing it into Blender to make a material for it.  It seems as though the calculations for generating the involute gear's x- and y-dimensions are probably incorrect.  I will provide information of what I get when importing to Blender and what I am supposed to get below.  I will also provide the information of the numbers and stuff I input for the Involute Gear.

Involute Gear's Dimensions (latest version of Blender): 18.895 mm for X-dimension, 18.9704 mm for Y-dimension, and 5.63 mm for Z-dimension.

Correct Involuted Gear's Dimensions (latest version of Blender): 19 mm for X-dimension, 19 mm for Y-dimension, and 5.63 mm for Z-dimension.  I calculated the x- and y-dimensions using the formulas provided from the link about gear design down below.

Involute Gear information: Number of Teeth is 17.  Module is 1.00 mm (divided the pitch circle diameter by the number of teeth).  Pressure Angle is 20 degrees.  Addendum Coefficient is 1.00.  Dedendum Coefficient is 1.25.  Root Fillet Coefficient is 0.00.  Profile Shift Coefficient is 0.00.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

Here are links to where I got information relating to measuring parts of a gear:
https://khkgears.net/new/gear_knowledge/gear_technical_reference/calculation_gear_dimensions.html
https://khkgears.net/new/gear_knowledge/gear-design-procedure-in-practical-design/design-shapes-of-spur-gears.html

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-11-20:
If I'm not mistaken Blender works on meshes so the shape dimension depends on the mesh algorithm and precision you choose.

SlyCoop91, 2024-11-20:
> Convert solid to mesh and try to measure mesh in FreeCAD before export to Blender.

Okay, just did what you said.  I converted the gear into a mesh and used the measuring tool to do some measurements.  From what I found, it feels like some calculations are off after measuring distance between vertexes.  I still got the same number for the Y-dimension, so I'll probably get the same number I got for the X-dimension when measuring it as well.  Here is a picture of the Y-dimension measurement:
[image]

tarman3, 2024-11-20:
I think your measure is not correct. Your measure X and Y depends on the rotation angle of the gear around the Z axis.
But radius is always 19 mm.

[image]
[image]

tarman3, 2024-11-20:
Please note that your measurement line does not go through the center.
[image]

SlyCoop91, 2024-11-21:
> Please note that your measurement line does not go through the center. ![Screenshot_20241120_211935](https://private-user-images.githubusercontent.com/188672424/388233058-885ee6b7-636a-46df-ae07-4bb4c00eab48.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIxNTI3NjYsIm5iZiI6MTczMjE1MjQ2NiwicGF0aCI6Ii8xODg2NzI0MjQvMzg4MjMzMDU4LTg4NWVlNmI3LTYzNmEtNDZkZi1hZTA3LTRiYjRjMDBlYWI0OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMVQwMTI3NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMWE1MWRkNzEzM2JlOTYzNjIzMTJkODlkZWU4ZWYyYmYyYmJiOWNhY2VhMmU5ODQyNjJmMjNkMmE5ZDRkOGEwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.DqAyA__rfFANNBgUFiYNgqcgn4qgkkX79AmEsiFgidU)

Thanks for your input.  Finally imported it to be about the size I wanted it, but there one tiny thing I am confused about.  When checking a X-dimension of one vertex, I saw there is an extra 0.00058 mm to it.  Is there a way to get rid of that extra bit of millimeters, or should I just ignore it?

maxwxyz, 2024-11-21:
Feel free to reopen if there is an actual bug

closed 2024-11-21 by maxwxyz