FreeCAD/FreeCAD #17996: flatpak version not render CJK in Draft WB Text annotation

soiamsoNG, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

flatpak version can not render CJK in Draft WB Text annotation
same machine the appimage version render correctly.

### Full version info

```shell
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-11-20:
Flatpak is not maintained by the FreeCAD team. Appimage is the official Linux version. You may want to report this to whoever maintains the flatpak itself as it may be a dependency version issue.

luzpaz, 2024-11-20:
@adrianinsaval can you please transfer to https://github.com/flathub/org.freecad.FreeCAD  
TIA

adrianinsaval, 2024-11-20:
does this work in appimage?

obelisk79, 2024-11-20:
> the appimage version render correctly.

@adrianinsaval

soiamsoNG, 2024-11-20:
> @adrianinsaval can you please transfer to https://github.com/flathub/org.freecad.FreeCAD  
> TIA

already report there，appimage version render correctly

Himura2la, 2026-10-29:
reprodiced on a nixos package. the fix: https://github.com/NixOS/nixpkgs/pull/580217