FreeCAD/FreeCAD #17998: The chinese language font could use Microsoft Yahei UI instead of Simsun.

suyangzuo, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some of font in FreeCAD is Simsun which is ugly for both chinese and english language.

These two screenshots below is using simsun font which is ugly for chinese language.
[image]
[image]

This screenshot showed two different font in setup dialog.
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-11-20:
I believe your suggested font is subject to copyright and therefore not compatible for use in FreeCAD. Google has a couple of open source Simplified Chinese fonts. Perhaps this one would be adequate? 

https://fonts.google.com/noto/specimen/Noto+Sans+SC

xtemp09, 2024-11-20:
@obelisk79, _Microsoft YaHei_ (_UI_) are available since _Windows Vista_. [^1]

[^1]: https://learn.microsoft.com/en-us/typography/font-list/microsoft-yahei

yorikvanhavre, 2024-11-20:
From https://learn.microsoft.com/en-us/typography/fonts/font-faq :

> Can I embed the fonts into a game, application or device I’m developing based on the document font embedding permissions?

> No, document font embedding permissions relate to embedding fonts in documents only, not embedding fonts in games, apps and devices.

and

> Apart from the document embedding rights described previously, you may not redistribute the Windows fonts. You may not copy them to other computers or servers, and you may not convert them to other formats, including bitmap formats, or modify them.

So unfortunately we are not permitted to ship that font within FreeCAD.

obelisk79, 2024-11-20:
> @obelisk79, _Microsoft YaHei_ (_UI_) are available since _Windows Vista_. [1](#user-content-fn-1-a1ecbfd69c20ebc653c0ef22075bd655)

FreeCAD supports multiple operating systems and any digital assets used also just have a compatible open source licence. YaHei funny doesn't meet this condition. Which is why I offered the Google font as a compatible option. Feel free to propose a better open source font. I do not know the Chinese language so I am limited.

obelisk79, 2024-11-20:
Solution must support Linux and MacOS. Not just Windows.

yorikvanhavre, 2024-11-20:
Isn't it possible for users to define the font used in Qt applications? On Linux it is, not sure about Windows...

xtemp09, 2024-11-20:
> Solution must support Linux and MacOS. Not just Windows.

> Isn't it possible for users to define the font used in Qt applications? On Linux it is, not sure about Windows...

It is possible to check the list of available fonts during start and choose appropriate.

suyangzuo, 2024-11-20:
### There are couple of options:
1. 阿里巴巴普惠体 3.0 `free for Commercial use`
[Official Page](https://www.alibabafonts.com/#/font)

2. 更纱黑体 `open source and free for Commercial use`
[Official Page](https://github.com/be5invis/Sarasa-Gothic)

3. 思源黑体 `open source and free for Commercial use`
[Official Page](https://github.com/adobe-fonts/source-han-sans)

And I have a question, if `Microsoft YaHei` is restricted by license, why texts in green frame could use it ?

suyangzuo, 2024-11-20:
> > So unfortunately we are not permitted to ship that font within FreeCAD.
> 
> There is no need to ship the font within _FreeCAD_, since it is available on _Windows_ platform by default.

That's right, `Microsoft Yahei` is contained in windows as default font.

suyangzuo, 2024-11-21:
> I believe your suggested font is subject to copyright and therefore not compatible for use in FreeCAD. Google has a couple of open source Simplified Chinese fonts. Perhaps this one would be adequate?
> 
> https://fonts.google.com/noto/specimen/Noto+Sans+SC

I think this could work.

luzpaz, 2024-11-22:
> > I believe your suggested font is subject to copyright and therefore not compatible for use in FreeCAD. Google has a couple of open source Simplified Chinese fonts. Perhaps this one would be adequate?
> > https://fonts.google.com/noto/specimen/Noto+Sans+SC
> 
> I think this could work.

Font [license](https://fonts.google.com/noto/specimen/Noto+Sans+SC/license):
```
Copyright 2014-2021 Adobe (http://www.adobe.com/), with Reserved Font Name 'Source'

This Font Software is licensed under the[ SIL Open Font License, Version 1.1 ](https://openfontlicense.org/open-font-license-official-text/). This license is copied below, and is also available with a FAQ at: https://openfontlicense.org/

SIL OPEN FONT LICENSE Version 1.1 - 26 February 2007
```

obelisk79, 2024-11-22:
It does *appear* to be compatible with lgpl.

https://fossa.com/blog/open-source-licenses-101-sil-open-font-license-ofl

[image]