FreeCAD/FreeCAD #17999: opening files / importing files - saving paths

Dirk-M, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello..
I have been working with versions 0.21.2 and 1.1.0dev for some time now.
As I mostly put together drawings from step files, I have a problem with the paths that the program saves.
When I open a drawing, its path is remembered and then displayed when importing.
If I import a step file via a different path, this is displayed again when I open a drawing.
Can't these different paths be saved in the background??

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * boltsfc 2022.11.5
  * btl 0.9.9
  * CADExchanger
  * dxf-library
  * fasteners 0.5.30
  * Help 1.0.3
  * InventorLoader 1.4.0
  * sheetmetal 0.5.5
  * symbols_library
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-20:
This is indeed annoying. There is only one `FileOpenSavePath`.

Dirk-M, 2024-11-20:
That explains a lot...
But it can definitely be adjusted...