FreeCAD/FreeCAD #18000: Stylesheets: Depressing toolbar buttons increases toolbar size (Light and Dark theme)

Roy-043, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When depressing buttons in the Draft snap toolbar, the toolbar becomes longer. It looks a bit strange. Both the Light and the Dark theme are affected. The Classic theme does not have this issue.

1. Toggle most Draft snaps off.
2. Restart FreeCad.
3. Toggle most Draft snaps on.

https://github.com/user-attachments/assets/9affaa84-637c-4b63-a933-dc975febcdf8

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods:
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-20:
Does "Lock toolbars" change the behavior?

Roy-043, 2024-11-21:
It does not matter if the toolbars are locked. If there is another toolbar to the right of the Draft snap toolbar, that toolbar will move. I suspect it is an issue with margins/padding.

Roy-043, 2024-11-27:
@Syres916 That works for me. Thank you.