FreeCAD/FreeCAD #18003: SubtractivePipe Fails

povilas-ohd, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It fails to do a SubtractivePipe operation is the sketch has more than one closed loops. Like if it has two holes. or Like in the example picture 

It says "SubtractivePipe001: Tool shape is not valid for boolean operation"
The issue persists in safe mode too.

It was perfectly working in freecad 0.21

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * CustomExporter
  * ThreadProfile 1.92.0
  * Curves 0.6.51
  * Ship 2024.10.23
  * Assembly4 0.50.2
  * dodo 1.0.1
  * STEMFIE 0.2.4
  * kicadStepUpMod 11.3.9
  * LCInterlocking
  * lattice2 1.0.0
  * ExplodedAssembly
  * CurvedShapes 1.0.12
```

### Subproject(s) affected?

PartDesign

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-20:
Can you provide a demo file ans steps how to reproduce.
@AIRCAP FYI

povilas-ohd, 2024-11-20:
[image]
[image]
![2024-11-20_13-00](https://github.com/user-attachments/assets/d951cb55-d65d-4639-b311-c0ff

Just like in pictures. Basic operation of subtractive pipe, in a sketch with single square on the edge works, but if I add another one at the bottom it doesn't .

I am using appimage version for freecad.

[spipe_bug.zip](https://github.com/user-attachments/files/17830208/spipe_bug.zip)
1cb80577)

povilas-ohd, 2024-11-21:
Example file in the zipped attachment.

maxwxyz, 2024-11-22:
I think pipe/lof with multiple profiles is currently not supported, right @kadet1090 ?

povilas-ohd, 2024-11-22:
> I think pipe/loft with multiple profiles is currently not supported, right @kadet1090 ?

But it worked perfectly with 0.21, the way I noticed as many previous files got broken while trying to recompute to new version.

kadet1090, 2024-11-22:
> I think pipe/loft with multiple profiles is currently not supported, right @kadet1090 ?

It should be, we had not supported stuff that resulted in multiple solids but for subtractive features it was allowed. I suspect that this has to do with toponaming changes. Can take look at this next week but not sooner probably.

maxwxyz, 2024-11-22:
@CalligaroV FYI

CalligaroV, 2024-11-24:
@maxwxyz thanks for the ping.

@povilas-ohd, while I too can confirm the issue, there's a thing I don't understand:

> It was perfectly working in freecad 0.21

I tried to reproduce with
```ruby
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * sheetmetal 0.5.5
  * Beltrami 1.1.0
  * kicadStepUpMod 11.4.1
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * Assembly4 0.50.15
  * Render 2024.9.25
  * Curves 0.6.51
  * freecad.gears 1.3.0
``` 

But I always get errors, both with [spipe_bug.zip](https://github.com/user-attachments/files/17830208/spipe_bug.zip) or recreating the model from scratch.

[image]

Can you please upload a file **made with v0.21** that works?
Without that it's difficult for me to move on with the analysis.

TIA!

povilas-ohd, 2024-11-24:
@CalligaroV Hi, 
Here is the file and and screenshot of exact body with functioning subtractive pipe that works but breaks in freecad 1.0

[honeycomb_wall_V1.zip](https://github.com/user-attachments/files/17893165/honeycomb_wall_V1.zip)

[image]

But I agree as with that test piece with cubes and two sketches it breaks for me too, but this one I provided works for some reason. Not sure what's the difference between the two.

CalligaroV, 2024-11-24:
Many thanks @povilas-ohd for the added info and the new test file!

I'll test it ASAP (most likely in the next hours) and give you a feedback.

Thanks again!

CalligaroV, 2024-11-24:
@povilas-ohd, I finished the basic tests on [honeycomb_wall_V1.zip](https://github.com/user-attachments/files/17893165/honeycomb_wall_V1.zip).
First of all I confirm that the issue happens in v1.0
```ruby
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * sheetmetal 0.5.5
  * Beltrami 1.1.0
  * kicadStepUpMod 11.4.1
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * Assembly4 0.50.15
  * Render 2024.9.25
  * Curves 0.6.51
  * freecad.gears 1.3.0
``` 
while it doesn't happen in v0.21
```ruby
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * sheetmetal 0.5.5
  * Beltrami 1.1.0
  * kicadStepUpMod 11.4.1
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * Assembly4 0.50.15
  * Render 2024.9.25
  * Curves 0.6.51
  * freecad.gears 1.3.0
``` 

I tried also with LS3
```ruby
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: a9810d509a6f112b5ac03d4d4831b67e6bffd5b7
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * sheetmetal 0.5.5
  * Beltrami 1.1.0
  * kicadStepUpMod 11.4.1
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * Assembly4 0.50.15
  * Render 2024.9.25
  * Curves 0.6.51
  * freecad.gears 1.3.0
``` 
and, while in this version/fork no error triggers about the SubtractivePipe, the output is different compared to v0.21.2 and, IMHO, wrong:
The SubtractivePipe looks computed but no "subtraction" is performed

v0.21
[image]

LS3
[image]

In other words, even if missing code is imported from LS3, further work is needed to restore 100% correctly this part of the model in v1.0.

Besides that, @maxwxyz, I'm not sure if this can be considered a Toponaming issue:
most likely the regression is caused by code imported for the integration but, as wrote above, also LS3 doesn't restores 100% correctly this part of the model and therefor some kind of fix should be developed regardless for the Toponaming integration.

FEA-eng, 2025-09-14:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```